bmax121/APatch #5: 安卓13 ，5.15，操作成功，但无法开机，需要提供什么吗

jxwdsb, 2023-08-27:
安卓13 ，5.15，操作成功，但无法开机，需要提供什么吗
./kptools -p Image --kpimg /root/tools/build/kpimg --skey 123456 --out kernel

[+] kptools image size 0x02700a00
[-] kptools kernel patch image size: 0x000255b4
[+] kernel image_size: 0x02700a00
[+] kernel uefi header: true
[+] kernel load_offset: 0x00000000
[+] kernel kernel_size: 0x027a0000
[+] kernel page_shift: 12
[+] kptools kernel new size 0x027c55b4
[+] kernel linux_banner offset: 0x01ad7e92
[+] kernel linux_banner: Linux version 5.15.94-android13-8-gdf77205a33ef (build-user@build-host) (Android (8508608, based on r450784e) clang version 14.0.7 (https://android.googlesource.com/toolchain/llvm-project 4c603efb0cca074e9238af8b4106c30add4418f6), LLD 14.0.7) #1 SMP PREEMPT Fri Aug 18 23:59:44 UTC 2023
[+] kernel version major: 5, minor: 15, patch: 94 
[+] kallsyms kallsyms_token_table offset: 0x019d78e8
[+] kallsyms endian: little
[+] kallsyms kallsyms_token_index offset: 0x019d7c20
[-] kallsyms find arm64 relocation table error
[+] kallsyms kallsyms_markers range: [0x019d6e18, 0x019d78e8), count: 0x000002b4
[+] kallsyms approximate kallsyms_offsets range: [0x015791f8, 0x01625fc4) count: 0x0002b373
[+] kallsyms kallsyms_names offset: 0x01625fd8
[+] kallsyms kallsyms_num_syms offset: 0x01625fd0, value: 0x0002b373
[+] kallsyms linux_banner index: 0x0001c5d9
[+] kallsyms kallsyms_offsets offset: 0x015791f8
[+] kptools map_start: 0x200, max_size: 0x800
[+] kallsyms kallsyms_lookup_name: type: T, offset: 0x00241674
[-] kallsyms can't find symbol: printk
[+] kallsyms paging_init: type: T, offset: 0x0238297c
[+] kallsyms memblock_reserve: type: T, offset: 0x003d3400
[+] kallsyms memblock_alloc_try_nid: type: T, offset: 0x023979f4
[+] kallsyms memstart_addr: type: D, offset: 0x01d0baf0
[+] kallsyms vabits_actual: type: D, offset: 0x02700010
[+] kallsyms kimage_voffset: type: D, offset: 0x01d0baf8
[+] kptools supercall key: 123456
[+] kptools patch done: kernel

bmax121, 2023-08-27:
printk 没有找到很奇怪，什么手机，boot.img或者kernel提供一下吧

jxwdsb, 2023-08-27:
> printk 没有找到很奇怪，什么手机，boot.img或者kernel提供一下吧

https://dl.google.com/android/gki/gki-certified-boot-android13-5.15-2023-04_r1.zip?hl=zh-cn

android13-5.15.94

bmax121, 2023-08-28:
5.15 以后确实没有 printk，但是有 _printk，可能是这个原因导致的，更新了，你再试一下

jxwdsb, 2023-08-28:
> 5.15 以后确实没有 printk，但是有 _printk，可能是这个原因导致的，更新了，你再试一下

仍然无法开机

[+] kptools image size 0x02700a00
[-] kptools kernel patch image size: 0x000255b4
[+] kernel image_size: 0x02700a00
[+] kernel uefi header: true
[+] kernel load_offset: 0x00000000
[+] kernel kernel_size: 0x027a0000
[+] kernel page_shift: 12
[+] kptools kernel new size 0x027c55b4
[+] kernel linux_banner offset: 0x01ad7e92
[+] kernel linux_banner: Linux version 5.15.94-android13-8-gdf77205a33ef (build-user@build-host) (Android (8508608, based on r450784e) clang version 14.0.7 (https://android.googlesource.com/toolchain/llvm-project 4c603efb0cca074e9238af8b4106c30add4418f6), LLD 14.0.7) #1 SMP PREEMPT Fri Aug 18 23:59:44 UTC 2023
[+] kernel version major: 5, minor: 15, patch: 94 
[+] kallsyms kallsyms_token_table offset: 0x019d78e8
[+] kallsyms endian: little
[+] kallsyms kallsyms_token_index offset: 0x019d7c20
[-] kallsyms find arm64 relocation table error
[+] kallsyms kallsyms_markers range: [0x019d6e18, 0x019d78e8), count: 0x000002b4
[+] kallsyms approximate kallsyms_offsets range: [0x015791f8, 0x01625fc4) count: 0x0002b373
[+] kallsyms kallsyms_names offset: 0x01625fd8
[+] kallsyms kallsyms_num_syms offset: 0x01625fd0, value: 0x0002b373
[+] kallsyms linux_banner index: 0x0001c5d9
[+] kallsyms kallsyms_offsets offset: 0x015791f8
[+] kptools map_start: 0x200, max_size: 0x800
[+] kallsyms kallsyms_lookup_name: type: T, offset: 0x00241674
[-] kallsyms can't find symbol: printk
[+] kallsyms _printk: type: T, offset: 0x001d2434
[+] kallsyms paging_init: type: T, offset: 0x0238297c
[+] kallsyms memblock_reserve: type: T, offset: 0x003d3400
[+] kallsyms memblock_alloc_try_nid: type: T, offset: 0x023979f4
[+] kallsyms memstart_addr: type: D, offset: 0x01d0baf0
[+] kallsyms vabits_actual: type: D, offset: 0x02700010
[+] kallsyms kimage_voffset: type: D, offset: 0x01d0baf8
[+] kptools supercall key: 112233
[+] kptools patch done: kernel

bmax121, 2023-08-29:
有时间的话加按照 readme 里面的，加 https://t.me/bmax121 一起调一下

closed 2023-12-26 by bmax121