bmax121/APatch #42: Failed to patch (polaris a13 pe+)

IPaffi, 2024-01-05:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

After flashing Boot my phone does not boot into firmware

### Device info

- Device: polaris
- OS Version: Pixel Experience (android 13, October patch)
- APatch Version: lasted (10302-7-gc3b7520)
- Kernel Version: 4.9.337

IPaffi, 2024-01-05:
Patch
APatch Boot Image Patcher
System is arm64
- Target image: /data/user/0/me.bmax.apatch/patch/boot.img
- Unpacking boot image
-
Patching
kernel
[+] kptools version: 803
[+] kptools image size 0x0257d018
[+] kptools kernel patch image size: 0x00027380
[+] kptools kpimg version: 803
[+] kptools kpimg compile time: 01:41:44 Jan 4 2024
[+] kernel image_size: 0x0257d018
[+] kernel uefi header: false
[+] kernel load_offset: 0x00080000
[+] kernel kernel_size: 0x02d36000
[+] kernel page_shift: 12
[+] kptools kernel
new
size 0x02d5d380
[+] kallsyms linux_banner 1: Linux version
4.9.337-Etude-Op.13-No.2-gdce351ed (nobody@android-build) (Prelude
clang version 16.0.2 ( https://github.com/llvm/llvm-project.git
18ddebe1a1a9bde349441631365f0472e9693520), LLD 16.0.2) #6
SMP
PREEMPT
Mon Sep 25
11:48:21
CEST
2023
[+] kallsyms linux_banner offset: 0x1d27e15
[+] kernel version major: 4, minor: 9, patch: 255
[+] kallsyms kallsyms_token_table offset: 0x01a0e400
[+] kallsyms endian: little
[+]
kallsyms kallsyms_token_index offset: 0x01a0e800
[?] kallsyms can't find arm64 relocation table
[+] kallsyms kallsyms_markers range: [0x01a0d300, 0x01a0e3c8), count:
0x00000219
[+] kallsyms approximate kallsyms_offsets range: [0x017b33e0,
0x01839520) count: 0x00021850
[+]
kallsyms kallsyms_names offset: 0x01839800
[+] kallsyms kallsyms_num_syms
offset: 0x01839700, value: 0x00021848
[+] kallsyms names table linux_banner index:
0x0000f12d
[+] kallsyms sure linux_banner index: 0
[+] kallsyms kallsyms_offsets offset: 0x017b3400
[+] kallsyms tcp_init_sock: type:
T, offset: 0x014ca040
[+] kptools map_start: 0x14ca040, max_size: 0x800
[+] kallsyms kallsyms_lookup_name: type: T, offset: 0x0012aef0
[+] kallsyms printk: type: T, offset: 0x000d85e0
[+] kallsyms paging_init: type: T, offset: 0x0218b200
[+] kallsyms memblock_reserve: type: T, offset: 0x00208600
[?] kallsyms no symbol: memblock_phys_alloc_try_nid
[+] kallsyms memblock_alloc_try_nid: type: T, offset: 0x021a6a00
[+] kallsyms memblock_mark_nomap: type: T, offset: 0x00208a80
[+]
kallsyms memstart_addr: type: D, offset: 0x01f8eb78
[?] kallsyms no symbol: vabits_actual
[+] kallsyms kimage_voffset: type: D, offset: 0x01f8eb88
[+] kptools supercall key: 0000
[+] kallsyms panic: type: T, offset: 0x0003e9d0
[+] kallsyms rest_init: type: t, offset: 0x016b7e80
[+] kallsyms cgroup_init: type: T, offset: 0x02199190
[+] kallsyms kernel_init: type: t, offset: 0x016b7f20
[?] kallsyms no symbol: report_cfi_failure
[?] kallsyms no symbol: ＿cfi_slowpath_diag
[?] kallsyms no symbol: ＿cfi_slowpath
[+] kallsyms copy_process: type: t, offset: 0x0003b7e0
[+] kallsyms cgroup_post_fork: type: T, offset: 0x00138750
[?] kallsyms no symbol: ＿do_execve_file
[+] kallsyms do_execveat_common: type: t, offset: 0x0023c610
[?] kallsyms no symbol: do_execve_common
[?] kallsyms no symbol: avc_denied
[+] kallsyms avc_denied -> avc_denied.Ilvm.6915241654359634978: type: t,
offset: 0x0048afc0
[+] kallsyms slow_avc_audit: type: T, offset: 0x00489c70
[+] kallsyms input_handle_event: type: t, offset: 0x00aa91c0
[?] kallsyms no symbol: vfs_statx
[?] kallsyms no symbol: do_statx
[+] kallsyms vfs_fstatat: type: T, offset: 0x00238ca0
[?] kallsyms no symbol: do_faccessat
[+] kallsyms sys_faccessat: type: T, offset:
0x0022d190
[+] kptools patch done:
kernel
- Repacking boot image
Write patched boot.img was successful
Output file is written to
/storage/emulated/0/Download/apatch_10309_ridu_boot.img

marat2509, 2024-01-05:
Create an issue at https://github.com/bmax121/KernelPatch/

closed 2024-01-05 by marat2509