bmax121/APatch #50: Would it be a good idea to consider putting this app on F-Droid

d-RealMusic-b, 2024-01-06:
Magisk and KernelSU are both up on f-droid for download, so why not APatch? And here is [F-Droid's inclusion policy](https://f-droid.org/docs/Inclusion_Policy/), I think APatch is in line with their policy.

linsui, 2024-02-13:
KernelSU is not on F-Droid, at least not in the official repo. The Magisk build metadata for F-Droid is maintained by me. If APatch agrees to be included, I can help.

d-RealMusic-b, 2024-02-13:
> KernelSU is not on F-Droid, at least not in the official repo. The Magisk build metadata for F-Droid is maintained by me. If APatch agrees to be included, I can help.

I was mistaken, KernelSU is currently in the lzzyOnDroid F-Droid Repo. Thanks for your work, hopefully we'll see APatch on F-Droid too.

bmax121, 2024-03-08:
I agree, if it doesn't inconvenience you too much.

bmax121, 2024-03-08:
@linsui

linsui, 2024-03-08:
I'll have a try. Looks like we need to build magisk first for these libs in https://github.com/bmax121/APatch/tree/main/app/libs/arm64-v8a. But it seems [libmagiskpolicy.so](https://github.com/bmax121/APatch/blob/main/app/libs/arm64-v8a/libmagiskpolicy.so) is from a different version of magisk?

pomelohan, 2024-03-09:
> I'll have a try. Looks like we need to build magisk first for these libs in https://github.com/bmax121/APatch/tree/main/app/libs/arm64-v8a. But it seems [libmagiskpolicy.so](https://github.com/bmax121/APatch/blob/main/app/libs/arm64-v8a/libmagiskpolicy.so) is from a different version of magisk?

Yes, it was a modified version, so that magiskpolicy can compatible with APatch

linsui, 2024-03-09:
Where is the source code of the modified version?

pomelohan, 2024-03-10:
> Where is the source code of the modified version?

Wait for @bmax121 upload patiently.

Btw, should the modified magiskpolicy being built together with APatch? If true, we may need hack gradle to do that.

linsui, 2024-03-11:
> Btw, should the modified magiskpolicy being built together with APatch? If true, we may need hack gradle to do that.

It needs to be built from source. We can build it first then put it in the required location.

pomelohan, 2024-03-14:
> > Btw, should the modified magiskpolicy being built together with APatch? If true, we may need hack gradle to do that.
> 
> It needs to be built from source. We can build it first then put it in the required location.

Sorry for the delay and thanks for supporting APatch. We’re currently working on developing these days so didn’t reply on time. We will notify you once we finished that. ;)

bmax121, 2024-03-15:
The source is at https://github.com/bmax121/Magisk,
Mainly changed this file: https://github.com/bmax121/Magisk/blob/master/native/src/sepolicy/policy.hpp

linsui, 2024-03-15:
Which commit do you use? I didn't see any change on master branch.

bmax121, 2024-03-15:
It's ok now

linsui, 2024-03-18:
IIUC, APatch doesn't use ONDK to build the rust code. Then does it make sense to build the magisk libs with ONDK? I have to rebuild ONDK from source so if it's not needed maybe I can skip that step.

shuvashish76, 2024-05-14:
@bmax121 [Requested](https://gitlab.com/IzzyOnDroid/repo/-/issues/542) at IzzyOnDroid.
According to their scanner report

```
No offending libs found.

Dangerous flags:
----------------
* usesCleartextTraffic

Permissions:
------------
* android.permission.INTERNET
* android.permission.WRITE_EXTERNAL_STORAGE
* android.permission.READ_EXTERNAL_STORAGE
* android.permission.QUERY_ALL_PACKAGES
* me.bmax.apatch.DYNAMIC_RECEIVER_NOT_EXPORTED_PERMISSION

```

Need some clarification regarding `usesCleartextTraffic` and `QUERY_ALL_PACKAGES`

pomelohan, 2024-05-15:
> @bmax121 [Requested](https://gitlab.com/IzzyOnDroid/repo/-/issues/542) at IzzyOnDroid. According to their scanner report
>
> ```
> No offending libs found.
>
>
> Dangerous flags:
> ----------------
> * [...]

Hey shuvashish76 Thank you for your contribution.

`usesCleartextTraffic` is for APatch Module WebUI feature and `QUERY_ALL_PACKAGES` is being used for selecting SuperUsers Applications. Are those enough? Feel free to notify me if you need more clarification ;)

pomelohan, 2024-05-15:
> What is the downloaded compiler in https://github.com/bmax121/KernelPatch/blob/main/.github/workflows/build.yml used for? Is there any simple way to build those kernel patch binaries?

Used for compiling kpimg, no more simple way but use this compiler because other one may lead to kpimg broken.

shuvashish76, 2024-05-15:
@pomelohan Thanks. Please add Fastlane metadata as mentioned [here](https://gitlab.com/-/snippets/1895688).

linsui, 2024-05-15:
> Used for compiling kpimg, no more simple way but use this compiler because other one may lead to kpimg broken.

Where is this special gcc from? How can I build it?

shuvashish76, 2024-05-15:
> `usesCleartextTraffic` is for APatch Module WebUI feature 

`usesCleartextTraffic` is only used locally on-device (and not to reach out to the internet)? Is that ensured e.g. via a "network security police"?

pomelohan, 2024-05-22:
> > `usesCleartextTraffic` is for APatch Module WebUI feature
> 
> `usesCleartextTraffic` is only used locally on-device (and not to reach out to the internet)? Is that ensured e.g. via a "network security police"?

Sorry for the delay, busy those days...

Yes, `usesCleartextTraffic` is only used locally on-device. (Located on: `/data/adb/modules/<module>/webroot`, it is providing an interface for module to control itself). It wouldn't reach out to the internet otherwise module author do that.

linsui, 2024-05-22:
> Update: No need special gcc, only aarch64-elf gcc is required. But it's better to use the same gcc version as KernelPatch using.

Am I understand it correctly that I just need an aarch64-elf gcc for all the build related to kernel patch?

pomelohan, 2024-05-22:
> > Update: No need special gcc, only aarch64-elf gcc is required. But it's better to use the same gcc version as KernelPatch using.
> 
> Am I understand it correctly that I just need an aarch64-elf gcc for all the build related to kernel patch?

Yes, but it is recommended to use the same gcc version as KernelPatch CI using.

marat2509, 2024-05-23:
> Thanks! That only leaves the storage permissions then which could use a short explanation, too 😉

@bmax121

linsui, 2024-05-23:
> Yes, but it is recommended to use the same gcc version as KernelPatch CI using because KernelPatch Project is special, unlike other programs. Newer or old version may lead to issues because of optimizations.

Can I use [this package](https://packages.debian.org/bookworm/gcc-aarch64-linux-gnu)? Or I should build the aarch64-elf-gcc somehow? I'm not sure what the difference is.

pomelohan, 2024-05-23:
> > Thanks! That only leaves the storage permissions then which could use a short explanation, too 😉
> 
> @bmax121 @pomelohan

storage permissions are used to store patched boot image and module upgrading🌹

pomelohan, 2024-05-23:
> I thought both of them are GCC 12.2?

[image]

Sorry I didn’t notice the version. I thought it was OK if version is 12.2 xD

linsui, 2024-05-23:
> Sorry I didn’t notice the version. I thought it was OK if version is 12.2 xD

So I can just use the gcc-linux-gnu? That's great. So the elf target is not necessary?

linsui, 2024-05-23:
> Yes have a try, author said maybe working😂

OK. :joy:

marat2509, 2024-05-26:
Released to https://apt.izzysoft.de/fdroid/index/apk/me.bmax.apatch

linsui, 2024-05-26:
> Yes have a try, author said maybe working😂

https://gitlab.com/linsui/fdroiddata/-/jobs/6942666334 Does this mean that it doesn't work?

pomelohan, 2024-05-26:
> > Yes have a try, author said maybe working😂
> 
> https://gitlab.com/linsui/fdroiddata/-/jobs/6942666334 Does this mean that it doesn't work?

Yes, kpimg.elf is necessary. Can’t we use prebuilt toolchain like the one KernelPatch using?

linsui, 2024-06-19:
Finally I build the gcc from source. I got an apk but not sure if it works. Could you please downgrade the jdk version to 17? Currently newer jdk versions are only available in testing channels of debian and they are conflict with packages I need to install to build gcc. I can build the apk with jdk 17 so I thought newer jdk is not necessary. I didn't use ondk because it takes a long time to build it and it seems APatch doesn't use ondk for other libs. But it seems this makes the size of magisk binaries much larger. Not sure if it's important.

linsui, 2024-07-03:
@bmax121 @pomelohan What can I do now? Do I need to build the magisk libs with ONDK? Can you downgrade the jdk version?

PR #682 (2024-10-08): Update README.md
Hi,

This small PR adds a button to get your app on F-Droid. You can later add a button for Google Play the same way :
`[<img src="https://play.google.com/intl/en_us/badges/images/generic/en-play-badge.png"
      alt="Get it on Google Play"
      height="80">]()`

It also adds a link to get the latest APK from the Releases Section.

You can now eventually add the [fdroid](https://github.com/topics/fdroid) and/or [f-droid](https://github.com/topics/f-droid) tags that will appear on the right side of the repo.

close #50
closed 2024-10-08 by marat2509