bmax121/APatch #55: [Manager Improvement] Add support for Devices with arm64 kernel but the system is in arm32 mode.

ravindu644, 2024-01-07:
Both Apatch and KSU apps can't be installed on a arm32 system, it says "application not installed".

But most of the time, these devices are using an arm64 kernel. is it possible to make the app work in arm32 devices too?

I can even patch the boot from apatch for my arm32 Galaxy A01.

Yervant7, 2024-01-14:
Check your CPU machine if it is armv8l need to create new support in KernelPatch too

ravindu644, 2024-01-14:
> Check your CPU machine if it is armv8l need to create new support in KernelPatch too

Scamsung fu*ked this phone.

[image]

bmax121, 2024-01-17:
hi, bro.  That's not easy, all the libraries need to be compatible, and kpimg need to do some compat too

ravindu644, 2024-01-17:
> hi, bro. That's not easy, all the libraries need to be compatible, and kpimg need to do some compat too

It's ok. Understood😌😊

closed 2024-01-18 by bmax121