bmax121/APatch #58: 【Question】How does apatch handle OTA upgrade?

fatinghenji, 2024-01-07:
As the title asks, how does apatch handle OTAs? Can I keep apatch when I do an OTA upgrade? Or do I need to process and install apatch after each OTA upgrade?

marat2509, 2024-01-09:
> need to process and install apatch after each OTA upgrade?

symbuzzer, 2024-01-12:
Is this planning to add?

marat2509, 2024-01-12:
> Is this planning to add?

More likely no than yes

ponces, 2024-01-12:
A possibility would be to flash on the inactive slot as Magisk does on a/b devices...

ponces, 2024-01-12:
No possibility for non a/b devices besides uninstalling/installing again

symbuzzer, 2024-01-12:
Is not possible with an addon.d script too?

symbuzzer, 2024-01-13:
https://github.com/symbuzzer/kernel-ota-survival-addon.d-script

symbuzzer, 2024-01-14:
> https://github.com/symbuzzer/kernel-ota-survival-addon.d-script @marat2509

I added both old boot-only devices and new boot_a, boot_b devices support and succesfully tested on Mi 10T Pro (boot only) and Poco X3 GT (boot_ab) devices.

bmax121, 2024-03-08:
#265

closed 2024-03-08 by bmax121