bmax121/APatch #66: Kernel patching is not possible

thiago92nr24, 2024-01-09:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

I tried the patch with two boot images I have, but both got the same error. My device is Nokia 5.3.

### Screenshots

[image]
[image]

### Logs

[APatch_bugreport_2024-01-08_22_47.tar.gz](https://github.com/bmax121/APatch/files/13867860/APatch_bugreport_2024-01-08_22_47.tar.gz)

### Device info

- Device: Nokia 5.3
- OS Version: 13
- APatch Version:10318-2
- Kernel Version:4.14.190

sekaiacg, 2024-01-09:
upload boot.img or kernel image

thiago92nr24, 2024-01-09:
[boot.zip](https://github.com/bmax121/APatch/files/13872271/boot.zip)
[boot2.zip](https://github.com/bmax121/APatch/files/13872273/boot2.zip)

sekaiacg, 2024-01-09:
The problem didn't occur on my device, I think it was caused by low RAM on the device.
[image]

```
01-08 22:47:31.891  7037  7037 D APatch  : APatch Boot Image Patcher
01-08 22:47:31.938  7037  7037 D APatch  : System is arm64
01-08 22:47:31.938  7037  7037 D APatch  : - Target image: /data/user/0/me.bmax.apatch/patch/boot.img
01-08 22:47:31.938  7037  7037 D APatch  : - Unpacking boot image
01-08 22:47:32.202 13280 13280 I cr_A11yState: Enabled accessibility services: [com.irishin.buttonsremapper/.remapper.ButtonsAccessibilityService, io.github.sds100.keymapper/.system.accessibility.MyAccessibilityService]
01-08 22:47:32.203 13280 13280 I cr_A11yState: Running accessibility services: [io.github.sds100.keymapper/.system.accessibility.MyAccessibilityService]
01-08 22:47:32.203 13280 13280 I cr_A11yState: Will check again after 2000 milliseconds.
01-08 22:47:32.343  7037  7037 D APatch  : - Patching kernel
01-08 22:47:30.238     0     0 I lowmemorykiller: Killing 'earchbox:search' (30946) (tgid 30946), adj 975,
01-08 22:47:30.238     0     0 I         : to free 27152kB on behalf of 'kptools' (14932) because
01-08 22:47:30.238     0     0 I         : cache 822856kB is below limit 402560kB for oom score 606
01-08 22:47:30.238     0     0 I         : Free memory is -25268kB above reserved.
01-08 22:47:30.238     0     0 I         : Free CMA is 524kB
01-08 22:47:30.238     0     0 I         : Total reserve is 32000kB
01-08 22:47:30.238     0     0 I         : Total free pages is 15064kB
01-08 22:47:30.238     0     0 I         : Total file cache is 923624kB
01-08 22:47:30.238     0     0 I         : GFP mask is 0x54280ca
01-08 22:47:30.316     0     0 I oom_reaper: reaped process 30946 (earchbox:search), now anon-rss:0kB, file-rss:4kB, shmem-rss:1776kB
01-08 22:47:32.590 14932 14932 F libc    : Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x751b24f000 in tid 14932 (kptools), pid 14932 (kptools)
```
It is temporarily recommended that you use tools under other systems. kpimg uses kpimg-android.
link: https://github.com/bmax121/KernelPatch/releases/tag/0.8.3
tool: kptools-linux or kptools-mac
Please note: You need to unpack boot.img to obtain the kernel image and then patch it.

example:
`./kptools-linux -p boot.img --kpimg kpimg-android --skey 1234 kernel_patched`

thiago92nr24, 2024-01-09:
[image]
I tried on a Linux PC, this error appears

Franciiscodev, 2024-01-09:
-Unpack boot.img first,
-Patch kernel image
-Repack Boot.img xD

thiago92nr24, 2024-01-09:
How do I patch the kernel image? Which commands?

Franciiscodev, 2024-01-09:
> How do I patch the kernel image? Which commands?

./kptools -p kernel --skey "1234" --kpimg kpimg --out Patchedkernel

Like this

thiago92nr24, 2024-01-10:
[image]
I extracted the kernel with aik, I tried with this command, but it gives this error

Franciiscodev, 2024-01-10:
> [image]
> I extracted the kernel with aik, I tried with this command, but it gives this error

Check if your kernel image is gz-compressed, then unpack

thiago92nr24, 2024-01-10:
That worked. Now do I replace the kernel in gz and repack the boot.img?

sekaiacg, 2024-01-10:
> That worked. Now do I replace the kernel in gz and repack the boot.img?

yes.

pomelohan, 2024-03-05:
Close since problem has been successfully solved.

closed 2024-03-05 by pomelohan