bmax121/APatch #84: 魅族18s 10.3刷入后卡meizu

Zsflee, 2024-01-10:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

boot链接  https://www.123pan.com/s/xykDVv-YNLN.html
修补后  刷入卡meizu
[image]

### Screenshots

[image]
[image]
[image]

### Device info

- Device:魅族18s
- OS Version:flyme10.3
- APatch Version:10363
- Kernel Version:5.4.148

ravindu644, 2024-01-10:
Enter the superkey to activate root.

Zsflee, 2024-01-10:
> Enter the superkey to activate root.

Flashing in the boot directly causes the device to boot up and cannot access the system.

Zsflee, 2024-01-10:
> Enter the superkey to activate root.

The screenshot I took is a system that only entered after flashing back to the official boot

ravindu644, 2024-01-10:
> > Enter the superkey to activate root.
> 
> Flashing in the boot directly causes the device to boot up and cannot access the system.

touch issue ?

ravindu644, 2024-01-10:
> > Enter the superkey to activate root.
> 
> The screenshot I took is a system that only entered after flashing back to the official boot

oh

Zsflee, 2024-01-10:
Repairs are successful.
[image]
[image]
[image]

ravindu644, 2024-01-10:
Can you try to flash the patched vbmeta.img from magisk?

how : 

Make a tar file using your original boot.img and vbmeta, then patch those files from magisk. Then extract the magisk patched tar, grab the vbmeta.img and flash it.

bmax121, 2024-01-14:
Try newest version with KP >= 0.8.4 and APatch >= 10420

wanghlstar, 2024-01-15:
10428 did not work yet on meizu 18  with flyme10.3

@Zsflee  are you ok ?

Zsflee, 2024-01-17:
> 10428 did not work yet on meizu 18 with flyme10.3
> 
> @Zsflee are you ok ?

Still stuck on the first screen

Zsflee, 2024-01-17:
> Try newest version with KP >= 0.8.4 and APatch >= 10420

Still stuck on the first screen  10440

bmax121, 2024-01-19:
Can't fix without device. Maybe a future update can solve it

closed 2024-01-19 by bmax121