bmax121/APatch #100: 可以增加KernelSU上的默认卸载模块选项吗？

XMWML, 2024-01-11:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

只有给应用关闭移除模块选项才能生效，这让一些全局模块，比如hosts在没有关闭移除模块选项的应用中不生效，ksu有默认卸载模块的选项可以设置黑白名单模式

### To Reproduce

只有给应用关闭移除模块选项才能生效，一些全局模块，比如systemless hosts在没有关闭移除模块选项的应用中不生效

### Expected behavior

添加一个让模块对于所有应用生效的选项

### Screenshots

这是ksu的选项
[image]

### Device info

- Device:xiaomi mi 10
- OS Version:14
- APatch Version:10398
- Kernel Version:4.19.157

bmax121, 2024-03-06:
apatch doesn't support unmount in KSU now, exclude button is not effective now

bmax121, 2024-03-07:
#253

closed 2024-03-07 by bmax121