bmax121/APatch #92: We got bugs🐞

ravindu644, 2024-01-11:
https://github.com/bmax121/APatch/assets/126038496/24b41ac3-22af-4a55-b9f3-dc0535855ca3

when changing language, back button seems not working, also it make the app crashes. btw it works.
[image]

App icon is darker too. Expected color = dark blue.

alex193a, 2024-01-11:
I'm unable to reproduce the issue, it may be a device-specific issue. Do you have any logs you can provide?

https://github.com/bmax121/APatch/assets/8174240/d2db9262-4280-4995-bc05-143f5c9de24f

The icon was updated in pull request https://github.com/bmax121/APatch/pull/89

ravindu644, 2024-01-11:
> I'm unable to reproduce the issue, it may be a device-specific issue. Do you have any logs you can provide?
> 
>  Record_2024-01-11-10-03-38.mp4 
> The icon was updated in pull request #89, @Traaanaelle can you confirm that the chosen color is correct?

Okay, I'll provide logs. yes, should an issue with one ui.

ravindu644, 2024-01-11:
> I'm unable to reproduce the issue, it may be a device-specific issue. Do you have any logs you can provide?
> 
>  Record_2024-01-11-10-03-38.mp4 
> The icon was updated in pull request #89, @Traaanaelle can you confirm that the chosen color is correct?

[APatch_bugreport_2024-01-11_15_02.tar.gz](https://github.com/bmax121/APatch/files/13901191/APatch_bugreport_2024-01-11_15_02.tar.gz)

Traaanaelle, 2024-01-11:
> I'm unable to reproduce the issue, it may be a device-specific issue. Do you have any logs you can provide?
> Record_2024-01-11-10-03-38.mp4
> 
> The icon was updated in pull request #89, @Traaanaelle can you confirm that the chosen color is correct?

Yes, I chose to replace the blue background with green

closed 2024-03-07 by bmax121