bmax121/APatch #98: ./kptools segment fault

CyanChanges, 2024-01-11:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

Segment fault while patching the boot
No new boot generated 

### To Reproduce

1. Enter the app,
2. Tap Patch and enter my passphrase and tap OK
3. Chose the image from a backup of KernelFlasher
4. Segment fault and no new boot generated 

### Expected behavior

Patch the image, and generate new-boot.img

### Screenshots

[image]

### Logs

[APatch_bugreport_2024-01-11_15_21.tar.gz](https://github.com/bmax121/APatch/files/13898691/APatch_bugreport_2024-01-11_15_21.tar.gz)

### Device info

- Device: Google Pixel 8 Pro
- OS Version: Android 14
- APatch Version: 10376-1-gc9312a1
- Kernel Version: 5.15.110

CyanChanges, 2024-01-11:
And the logs from LogFox
[crash--_kptools-11_01-15-19-55_286.zip](https://github.com/bmax121/APatch/files/13898739/crash--_kptools-11_01-15-19-55_286.zip)

bmax121, 2024-01-14:
Please goto KernelPatch and open a issue

closed 2024-01-14 by bmax121