bmax121/APatch #118: kallsyms kallsyms_num_syms error

chengnan049, 2024-01-12:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

RT

### To Reproduce

Extract `boot.img` from Fastboot ROM > Patch it in APatch APP > See error

### Screenshots

[image]

### Logs

[APatch_bugreport_2024-01-12_19_41.tar.gz](https://github.com/bmax121/APatch/files/13918643/APatch_bugreport_2024-01-12_19_41.tar.gz)

### Device info

- Device:Redmi Note12 5G (sunstone)
- OS Version:Android 13,MIUI 14 EEA Global
- APatch Version:10417
- Kernel Version:5.4.233-qgki-gfd9e0bcf9fc3

chengnan049, 2024-01-12:
If I use KernelPatch on macOS, will error to, logs are follows

./kptools-mac -p kernel-b --skey "1234" --kpimg kpimg-android --out kernel
[+] kptools version: 803
[+] kptools image size 0x02a7e200
[+] kptools kernel patch image size: 0x00027380
[+] kptools kpimg version: 803
[+] kptools kpimg compile time: 13:56:31 Jan 11 2024
[+] kernel image_size: 0x02a7e200
[+] kernel uefi header: true
[+] kernel load_offset: 0x00080000
[+] kernel kernel_size: 0x03048000
[+] kernel page_shift: 12
[+] kptools kernel new size 0x0306f380
[+] kallsyms linux_banner 1: Linux version 5.4.233-qgki-gfd9e0cbf9fc3 (builder@pangu-build-component-vendor-179427-zsc75-9wlcq-9h763) (Android (6877366 based on r383902b1) clang version 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79), LLD 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79)) #1 SMP PREEMPT Fri Nov 17 10:44:16 UTC 2023
[+] kallsyms linux_banner offset: 0x1c94fb8
[+] kernel version major: 5, minor: 4, patch: 233
[+] kallsyms kallsyms_token_table offset: 0x01987358
[+] kallsyms endian: little
[+] kallsyms kallsyms_token_index offset: 0x01987680
[+] kallsyms find arm64 relocation table range: [0x02589718, 0x0284bb40), text_va: 0xffffffc010000000, count: 0x0001d6d7
[+] kallsyms apply 0x000197d5 relocation entries
[+] kallsyms kallsyms_markers range: [0x01986880, 0x01987358), count: 0x000002b6
[+] kallsyms approximate kallsyms_offsets range: [0x0151bc54, 0x0154e004) count: 0x0000c8ec
[+] kallsyms kallsyms_names offset: 0x015c91e8
[-] kallsyms kallsyms_num_syms error
[-] kptools analyze_kallsym_info error

[image]

chengnan049, 2024-01-12:
Same issuse https://github.com/bmax121/KernelPatch/issues/33

closed 2024-01-13 by chengnan049

marat2509, 2024-01-13:
> If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch

chengnan049, 2024-02-04:
> > If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch

It still didn't work, I found the same issues but no one responsed

engharat, 2025-01-01:
It has been fixed on KernelPatch, any hope to get it fixed in Apatch too?
[https://github.com/bmax121/KernelPatch/issues/12](url)