bmax121/APatch #120: Google Pixel 7 刷入后自动恢复到官方内核

mdogwoop, 2024-01-12:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

如标题

### Device info

- Device: Google Pixel 7
- OS Version:14 (UQ1A.240105.004)
- APatch Version:10417-1-g1f1ff3e
- Kernel Version:5.10.177-android13-4-00003-ga7208022a7ea-ab10815828

marat2509, 2024-01-12:
Please clarify the problem: after upgrading or re-flashing the firmware, do you revert to the factory kernel? Please answer in English

mdogwoop, 2024-01-12:
> Please clarify the problem: after upgrading or re-flashing the firmware, do you revert to the factory kernel? Please answer in English

I don't understand what you want me to do?  Do I need to re-flash the firmware?

marat2509, 2024-01-12:
Please write the title in English. I guess I didn't understand correctly what the problem was.

mdogwoop, 2024-01-12:
> Please write the title in English. I guess I didn't understand correctly what the problem was.

I updated the title and I would like to see a reminder that English is required next time I create an issue

mdogwoop, 2024-01-12:
In my experience, installing Kernel SU or Magisk on this device is different from other phones. The installation method on the KSU support page is as follows https://kernelsu.org/guide/installation.html#patch-boot-  img-manually, and the installation of magisk requires patching init_boot.img

PA733, 2024-01-14:
Using Pixel 7 Pro. Can't reproduce the problem you're experiencing.
Apatch works find on my device.

AP11.231215.007
5.10.189

PA733, 2024-01-14:
You may need to reboot the system once after updating the firmware, then patch the kernel, then reboot to the bootloader and flash the patched kernel.
你可能要在更新固件后，重启一次系统，然后 patch 内核，再重启到 bootloader，最后刷入新内核。

burntoc, 2024-02-22:
> Using Pixel 7 Pro. Can't reproduce the problem you're experiencing. Apatch works fine on my device.
> 
Any chance you can share how you did that?  I have a P7P as well and Apatch will patch the boot.img but not init_boot.img which is what we need, I believe.  Trying to flash the patched boot.img to init_boot fails, as it should (thankfully).  I'm not clear on the steps even after having read the magiskboot stuff.