bmax121/APatch #121: After type Superkey, Redmi Note 11 5G(evergo) auto restart

Webpage-gh, 2024-01-13:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

<details>
  <summary>中文点这里</summary>
如标题，我在Patch原始映像（未经Magisk修补后的映像）并刷入后，手机可以正常使用。当我在APatch点击“超级密钥”并输入在修补时设定的超级密钥后，手机卡顿了半秒左右，接着黑屏，最后进入开机画面了。这个过程就像是软重启（system_server被结束）一样。

随后，为了排查问题，我尝试刷入了先在Magisk Patch，然后在Apatch Patch后的映像，便于调试问题。结果是Magisk正常工作，Apatch依然像是上文中提到的软重启一样。
</details>
As mentioned in the title, when I patch the original image (not pre-patched with Magisk) and flash it, the phone works fine. However, when I click on "SuperKey" in APatch and enter the super key set during the patching process, the phone lags for about half a second, then goes black, and finally enters the boot screen. This process resembles a soft restart (system_server being terminated).

Subsequently, to troubleshoot the issue, I attempted to flash an image pre-patched with Magisk and then patched with APatch for debugging purposes. The result was that Magisk worked correctly, but APatch still exhibited the soft restart issue mentioned earlier.

### To Reproduce

1. Flash the patched image, and boot up.
2. Open APatch, click on "Supercall Key."
3. Enter the designated key, click "Confirm."
4. Check for errors.

### Expected behavior

I'm not sure, but I presume that in the Kernel Patch or Android Patch sections, unknown items should no longer appear. Root access can be granted successfully, and AP modules or KP modules can be used.

### Logs

Perhaps the [LSPosed log](
https://github.com/bmax121/APatch/files/13926881/LSPosed_2024-01-13T13_23_38.394671.zip) might be helpful for you. The log from [here](https://github.com/bmax121/APatch/files/13926884/APatch_bugreport_2024-01-13_13_34.tar.gz) is from Apatch Manager - Settings - Report log.

### Device info

- Device: Redmi Note 11 5G (evergo)
- OS Version: MIUI 14.0.4.0(TGBINXM), Android 13 TP1A.220624.014
- APatch Version: 10417-1-g1f1ff3e (10418) ([Github Release](https://github.com/bmax121/APatch/releases/tag/10418))
- Kernel Version: 4.14.186-perf-ga3328744cf39

### Additional context

<details>
  <summary>中文点这里</summary>
关于第二个检查项，我Patch Boot映像成功了，但没有依靠KernelPatch拿到root权限，修补后的映像（无论是原始映像还是Magisk事先修补后的映像）都可以正常启动设备。

关于第四个检查项，我在[KernelPatch的Issue](https://github.com/bmax121/KernelPatch/issues/19)发现了疑似同机型的议题，不过Redmi Note 11系列有evergo（5G版）、selenes（国内的4G版）、spes（印度版）和spesn（欧版），他们代号不同，因此内核可能有所区别。

我可以提取boot.img和/proc/kallsyms中的内容，如果您需要的话。
</details>
Regarding the second check item, I successfully patched the Boot image, but did not gain root access relying on KernelPatch. The patched image, whether it's the original image or the one pre-patched with Magisk, can boot the device normally.

For the fourth check item, I found a similar issue for the same device model on [KernelPatch's Issue](https://github.com/bmax121/KernelPatch/issues/19). However, the Redmi Note 11 series has different codenames such as evergo (5G version), selenes (Chinese 4G version), spes (Indian version), and spesn (European version), suggesting potential kernel variations.

I can extract the content of boot.img and /proc/kallsyms if needed.

bmax121, 2024-01-14:
Here is steps to get kernel log
1. Root your devices with kernelsu or magisk first. (to get log)
2. Patch boot.img which you used in the previous step and then flash it
3. Open a root shell and execute 'cat /proc/kmsg'
4. Type Superkey, Make your devices crash and reboot. And save logs of previous step (log1)
5. After reboot, Save all file in /sys/fs/pstore/  (log2)
6. Upload log1 and log2

bmax121, 2024-01-14:
If it is convenient, you'd better execute 'dmesg' to get kerne boot log after the device restarts **immediately**.
This is log3。

Upload log1, log2, log3,

Webpage-gh, 2024-01-14:
Thank you for your response. I have tried to extract **these logs
[kernellog.zip](https://github.com/bmax121/APatch/files/13931953/kernellog.zip)** following your steps as much as possible. When extracting log1, my steps were as follows:

1. Open Termux and execute `tsu` to enter the root shell.
2. Execute the command `while true; do cat /proc/kmsg >> /sdcard/log1.txt; done` (just executing `cat /proc/kmsg` alone won't save log1 for submission).
3. Switch to APatch, input the SuperKey to trigger the crash.

However, after the reboot, I only managed to extract log2 and log3, and log1 did not appear. Even though I did observe the generation of log1 before the restart. I tried this process twice, and the results were the same. I hope these logs help you investigate the issue.

bmax121, 2024-01-17:
Is there any logs in /data/adb/ap/log

bmax121, 2024-01-17:
And upload your boot.img, I'll patch it for you to test

Webpage-gh, 2024-01-17:
Below is my split compressed file, including boot.img and syms.txt. Due to GitHub's restrictions on file size and format, this is the only way to upload files in this issue. For others participating in the issue, it's challenging for us to achieve perfection. Please forgive our mistakes and don't be upset. We just want to solve this problem in a friendly and efficient manner.
[boot.7z.001.zip](https://github.com/bmax121/APatch/files/13965295/boot.7z.001.zip) and 
[boot.7z.002.zip](https://github.com/bmax121/APatch/files/13965320/boot.7z.002.zip)

Webpage-gh, 2024-01-18:
Sorry. I can't find folder /data/adb/ap
[image]

chen2021-web, 2024-01-28:
> And upload your boot.img, I'll patch it for you to test

I have also use evergo.And  I have same problem.I will compress boot.img and transfer it to the cloud storage. The link is as follows:https://wwd.lanzout.com/irPG61mjblni

teteko233, 2024-01-28:
After android crash,  `cat /proc/kmsg > log1.txt` doesn't work and nothing will be output. Then I extract log from `/proc/last_kmsg` .

My rom is [evo x](https://github.com/MTK-DM-810-UNIFIED/official_devices/releases/tag/evox-7.9.10-eol) and I use [boot.img](https://github.com/MTK-DM-810-UNIFIED/official_devices/releases/download/evox-7.9.10-eol/boot.img) of this rom.

Here are my steps.

1. use apatch patches the boot.img which has patched by magisk delta.
1. type `adb shell`  and  `su` to enter the root shell.
2. Type Superkey in android device. Make it crash and reboot.
3. type below code to get logs.

```
cat /proc/last_kmsg > /storage/emulated/0/log/log1.txt
cp -r /sys/fs/pstore /storage/emulated/0/ 
dmesg > /storage/emulated/0/log/log3.txt
```

4. copy logs from my device to PC. This is my [log.zip](https://github.com/bmax121/APatch/files/14076001/log.zip)

bmax121, 2024-03-08:
Try the latest version, and if it still doesn't work, reopen the issue.

closed 2024-03-08 by bmax121

Webpage-gh, 2024-03-09:
[image]
Indeed, thanks to everyone who contributed to this.

KateWasHere001, 2024-06-18:
it seems A11 boot.img is not suitable yet.

[boot.7z.001.zip](https://github.com/user-attachments/files/15880946/boot.7z.001.zip)
[boot.7z.002.zip](https://github.com/user-attachments/files/15880947/boot.7z.002.zip)

Webpage-gh, 2024-06-19:
What is A11

KateWasHere001, 2024-07-29:
> What is A11

Android 11