bmax121/APatch #123: APatch not working on Samsung devices

BlackMesa123, 2024-01-13:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

The stock kernel image patched with APatch doesn't boots. Manual patching the kernel image with KernelPatch was also tested.

### To Reproduce

Patch the boot.img via the APatch app and flash the newly generated boot.img via a custom recovery.

### Expected behavior

Device should boot and function as normal.

### Logs

last_kmsg file of the boot.img patched via app: [last_kmsg-app.zip](https://github.com/bmax121/APatch/files/13928146/last_kmsg-app.zip)
last_kmsg file of the boot.img patched manually: [last_kmsg-manual.zip](https://github.com/bmax121/APatch/files/13928147/last_kmsg-manual.zip)

For some strange reason, no kernel kmsg output is visible. If you have another way to obtain logs please let me know.

### Device info

- Device: Samsung Galaxy A52s 5G
- OS Version: Android 14 (One UI 6.0)
- APatch Version: 10419 (KernelPatch 0.8.3)
- Kernel Version: Samsung Yupik 5.4.233 stock kernel, based off Qualcomm LA.UM.9.14.r1-22000-LAHAINA.QSSI12.0 (https://github.com/BlackMesa123/android_kernel_samsung_sm7325)

### Additional context

Original untouched kernel image file: [Image-stock.zip](https://github.com/bmax121/APatch/files/13928159/Image-stock.zip)
Patched kernel image file: [Image_patched.zip](https://github.com/bmax121/APatch/files/13928160/Image_patched.zip)
<details>
<summary>kptools output:</summary>

```console
❯ ./kptools-linux -p Image --kpimg kpimg-android --skey test
[+] kptools version: 803
[+] kptools image size 0x02e8ba00
[+] kptools kernel patch image size: 0x00027380
[+] kptools kpimg version: 803
[+] kptools kpimg compile time: 10:55:17 Jan 13 2024
[+] kernel image_size: 0x02e8ba00
[+] kernel uefi header: true
[+] kernel load_offset: 0x00080000
[+] kernel kernel_size: 0x033e4000
[+] kernel page_shift: 12
[+] kptools kernel new size 0x0340b380
[+] kallsyms linux_banner 1: Linux version 5.4.233-qgki-27763529-abA528BXXS5FWL4 (dpi@VPHLR1014) (Android (6877366 based on r383902b1) clang version 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79), LLD 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79)) #1 SMP PREEMPT Thu Dec 7 14:49:11 +07 2023
[+] kallsyms linux_banner offset: 0x2080038
[+] kallsyms linux_banner 2: Linux version 5.4.233-qgki-27763529-abA528BXXS5FWL4 (dpi@VPHLR1014) (Android (6877366 based on r383902b1) clang version 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79), LLD 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79)) #1 SMP PREEMPT Thu Dec 7 14:49:11 +07 2023
[+] kallsyms linux_banner offset: 0x240d784
[+] kernel version major: 5, minor: 4, patch: 233 
[+] kallsyms kallsyms_token_table offset: 0x01fb69c8
[+] kallsyms endian: little
[+] kallsyms kallsyms_token_index offset: 0x01fb6cf0
[+] kallsyms find arm64 relocation table range: [0x02888b58, 0x02b9b2c0), text_va: 0xffffffc010000000, count: 0x00020c4f
[+] kallsyms apply 0x0001cd31 relocation entries
[+] kallsyms kallsyms_markers range: [0x01fb5e28, 0x01fb69c8), count: 0x000002e8
[+] kallsyms approximate kallsyms_offsets range: [0x01afc5ec, 0x01bb64a4) count: 0x0002e7ae
[+] kallsyms kallsyms_names offset: 0x01bb64b8
[+] kallsyms kallsyms_num_syms offset: 0x01bb64b0, value: 0x0002e7ad
[+] kallsyms names table linux_banner index: 0x0001b4c3
[+] kallsyms sure linux_banner index: 1
[+] kallsyms kallsyms_offsets offset: 0x01afc5f0
[+] kallsyms tcp_init_sock: type: T, offset: 0x0147b8b0
[+] kptools map_start: 0x147b8b0, max_size: 0x800
[+] kallsyms kallsyms_lookup_name: type: T, offset: 0x00376cf4
[+] kallsyms printk: type: T, offset: 0x0031fde8
[+] kallsyms paging_init: type: T, offset: 0x0270b78c
[+] kallsyms memblock_reserve: type: T, offset: 0x004a211c
[+] kallsyms memblock_phys_alloc_try_nid: type: T, offset: 0x027255d8
[+] kallsyms memblock_mark_nomap: type: T, offset: 0x004a23d0
[+] kallsyms memstart_addr: type: D, offset: 0x02547440
[+] kallsyms vabits_actual: type: D, offset: 0x026056b0
[+] kallsyms kimage_voffset: type: D, offset: 0x02547460
[+] kptools supercall key: test
[+] kallsyms panic: type: T, offset: 0x002917c8
[+] kallsyms rest_init: type: T, offset: 0x0170f614
[+] kallsyms cgroup_init: type: T, offset: 0x02718cec
[?] kallsyms no symbol: kernel_init
[?] kallsyms no symbol: report_cfi_failure
[?] kallsyms no symbol: __cfi_slowpath_diag
[+] kallsyms __cfi_slowpath: type: T, offset: 0x0041b1a0
[+] kallsyms copy_process: type: t, offset: 0x0028e2d0
[+] kallsyms cgroup_post_fork: type: T, offset: 0x0038175c
[+] kallsyms __do_execve_file: type: t, offset: 0x004e32a8
[?] kallsyms no symbol: do_execveat_common
[?] kallsyms no symbol: do_execve_common
[+] kallsyms avc_denied: type: t, offset: 0x00719818
[+] kallsyms slow_avc_audit: type: T, offset: 0x007188cc
[+] kallsyms input_handle_event: type: t, offset: 0x00c937dc
[+] kallsyms vfs_statx: type: T, offset: 0x004df7d0
[?] kallsyms no symbol: do_statx
[?] kallsyms no symbol: vfs_fstatat
[+] kallsyms do_faccessat: type: T, offset: 0x004d3298
[?] kallsyms no symbol: sys_faccessat
[+] kptools patch done: Image_patched
```

</details>

Stock kernel binaries can be downloaded here: https://github.com/BlackMesa123/proprietary_vendor_samsung_a52sxq/releases/tag/A528BXXS5FWL4_BTU

BlackMesa123, 2024-01-13:
Building a custom kernel image with disabled Samsung security drivers (https://github.com/BlackMesa123/android_kernel_samsung_sm7325/commit/7569eb2c1549a98f0edef3c2d22d92d2211aaa1d) solves this.

[image]<br>

Perharps extra patches are required to handle Samsung shipped kernels?

Custom build kernel image file: [Image.zip](https://github.com/bmax121/APatch/files/13928270/Image.zip)
Same but KernelPatch'd: [Image_patched.zip](https://github.com/bmax121/APatch/files/13928271/Image_patched.zip)

<details>
<summary>kptools output:</summary>

```console
❯ ./kptools-linux -p Image --kpimg kpimg-android --skey test
[+] kptools version: 803
[+] kptools image size 0x02a6a200
[+] kptools kernel patch image size: 0x00027380
[+] kptools kpimg version: 803
[+] kptools kpimg compile time: 10:55:17 Jan 13 2024
[+] kernel image_size: 0x02a6a200
[+] kernel uefi header: true
[+] kernel load_offset: 0x00080000
[+] kernel kernel_size: 0x02e84000
[+] kernel page_shift: 12
[+] kptools kernel new size 0x02eab380
[+] kallsyms linux_banner 1: Linux version 5.4.233-qgki-16133-g886f0b784ccc (mesa@salvo-z390aorusmaster) (Android (6877366 based on r383902b1) clang version 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79), LLD 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79)) #1 SMP PREEMPT Sat Jan 13 14:58:48 CET 2024
[+] kallsyms linux_banner offset: 0x22ff904
[+] kernel version major: 5, minor: 4, patch: 233 
[+] kallsyms kallsyms_token_table offset: 0x01f74780
[+] kallsyms endian: little
[+] kallsyms kallsyms_token_index offset: 0x01f74aa8
[?] kallsyms can't find arm64 relocation table
[+] kallsyms kallsyms_markers range: [0x01f73be8, 0x01f74780), count: 0x000002e6
[+] kallsyms approximate kallsyms_offsets range: [0x01abc56c, 0x01b75bb4) count: 0x0002e592
[+] kallsyms kallsyms_names offset: 0x01b75bc0
[+] kallsyms kallsyms_num_syms offset: 0x01b75bb8, value: 0x0002e590
[+] kallsyms names table linux_banner index: 0x0001b391
[+] kallsyms sure linux_banner index: 0
[+] kallsyms kallsyms_offsets offset: 0x01abc570
[+] kallsyms tcp_init_sock: type: T, offset: 0x0146b490
[+] kptools map_start: 0x146b490, max_size: 0x800
[+] kallsyms kallsyms_lookup_name: type: T, offset: 0x00375970
[+] kallsyms printk: type: T, offset: 0x0031ebcc
[+] kallsyms paging_init: type: T, offset: 0x025eb524
[+] kallsyms memblock_reserve: type: T, offset: 0x004a0d74
[+] kallsyms memblock_phys_alloc_try_nid: type: T, offset: 0x026052d4
[+] kallsyms memblock_mark_nomap: type: T, offset: 0x004a1028
[+] kallsyms memstart_addr: type: D, offset: 0x02435480
[+] kallsyms vabits_actual: type: D, offset: 0x024ed038
[+] kallsyms kimage_voffset: type: D, offset: 0x024354a0
[+] kptools supercall key: test
[+] kallsyms panic: type: T, offset: 0x00290410
[+] kallsyms rest_init: type: T, offset: 0x016fd2c4
[+] kallsyms cgroup_init: type: T, offset: 0x025f89e8
[?] kallsyms no symbol: kernel_init
[?] kallsyms no symbol: report_cfi_failure
[?] kallsyms no symbol: __cfi_slowpath_diag
[+] kallsyms __cfi_slowpath: type: T, offset: 0x00419e20
[+] kallsyms copy_process: type: t, offset: 0x0028d008
[+] kallsyms cgroup_post_fork: type: T, offset: 0x003803d8
[+] kallsyms __do_execve_file: type: t, offset: 0x004e1838
[?] kallsyms no symbol: do_execveat_common
[?] kallsyms no symbol: do_execve_common
[+] kallsyms avc_denied: type: t, offset: 0x00716e24
[+] kallsyms slow_avc_audit: type: T, offset: 0x00715ed8
[+] kallsyms input_handle_event: type: t, offset: 0x00c853c4
[+] kallsyms vfs_statx: type: T, offset: 0x004ddd8c
[?] kallsyms no symbol: do_statx
[?] kallsyms no symbol: vfs_fstatat
[+] kallsyms do_faccessat: type: T, offset: 0x004d17d8
[?] kallsyms no symbol: sys_faccessat
[+] kptools patch done: Image_patched
```

</details>

BlackMesa123, 2024-01-13:
By taking a look at [Magisk](https://github.com/topjohnwu/Magisk/blob/79334b77027cdb19bef3450181e8a1dd3d7c4fd2/scripts/boot_patch.sh#L222-L231), we can see Samsung kernel images are patched out to address a few issues:
```bash
# Remove Samsung RKP
./magiskboot hexpatch kernel \
49010054011440B93FA00F71E9000054010840B93FA00F7189000054001840B91FA00F7188010054 \
A1020054011440B93FA00F7140020054010840B93FA00F71E0010054001840B91FA00F7181010054 \
&& PATCHEDKERNEL=true

# Remove Samsung defex
# Before: [mov w2, #-221]   (-__NR_execve)
# After:  [mov w2, #-32768]
./magiskboot hexpatch kernel 821B8012 E2FF8F12 && PATCHEDKERNEL=true
```

Samsung RKP patch was originally made by [Chainfire](https://chainfire.eu/articles/950/SuperSU_v2.77_BETA_-_Note7_Exynos_shenanigans) back in the days for SuperSU to bypass [CONFIG_RKP_NS_PROT](https://github.com/exynos-linux-stable/dreamlte/blob/6e55204236228b0a06e63c64d43b20411995178c/fs/exec.c#L1181-L1187), this isn't a thing anymore in modern devices but it's still present on older ones. Notice this patch still doesn't works on every device (https://github.com/topjohnwu/Magisk/issues/7254).
Samsung Defex patch was implemented in 2018 (https://github.com/topjohnwu/Magisk/issues/426), like for RKP this isn't really necessary on newer devices as Defex will automatically be disabled when in [recovery boot mode/unlocked bootloader](https://github.com/BlackMesa123/android_kernel_samsung_sm7325/blob/e76fd1fdabdca7554e5d7df0dbd10dcbabdfb3fd/security/samsung/defex_lsm/core/defex_lsm.c#L68-L77).

ravindu644, 2024-01-13:
> Building a custom kernel image with disabled Samsung security drivers ([BlackMesa123/android_kernel_samsung_sm7325@7569eb2](https://github.com/BlackMesa123/android_kernel_samsung_sm7325/commit/7569eb2c1549a98f0edef3c2d22d92d2211aaa1d)) solves this.
> 
> <img alt="image" width="250" src="https://private-user-images.githubusercontent.com/13062958/296486189-98c4e766-9cfb-43ef-86fe-87eafde5c9e7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUxNTkwNzcsIm5iZiI6MTcwNTE1ODc3NywicGF0aCI6Ii8xMzA2Mjk1OC8yOTY0ODYxODktOThjNGU3NjYtOWNmYi00M2VmLTg2ZmUtODdlYWZkZTVjOWU3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEzVDE1MTI1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzZmZjZTk0ODJkODYyMjYwOTBmMjY3NmFiYmM0ZmVjM2FlNDcwNTE2Mzg4NTA0YjJkZjQwZGRjMjhlZDZiMzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RVTXqCAM7Qfe06J1JkLyXYcFEy8a3CBspVqywlXP2vI">
> 
> Perharps extra patches are required to handle Samsung shipped kernels?
> 
> Custom built kernel image file: [Image.zip](https://github.com/bmax121/APatch/files/13928270/Image.zip) Same but KernelPatch'd: [Image_patched.zip](https://github.com/bmax121/APatch/files/13928271/Image_patched.zip)
> 
> kptools output:
> ```
> ❯ ./kptools-linux -p Image --kpimg kpimg-android --skey test
> [+] kptools version: 803
> [+] kptools image size 0x02a6a200
> [+] kptools kernel patch image size: 0x00027380
> [+] kptools kpimg version: 803
> [+] kptools kpimg compile time: 10:55:17 Jan 13 2024
> [+] kernel image_size: 0x02a6a200
> [+] kernel uefi header: true
> [+] kernel load_offset: 0x00080000
> [+] kernel kernel_size: 0x02e84000
> [+] kernel page_shift: 12
> [+] kptools kernel new size 0x02eab380
> [+] kallsyms linux_banner 1: Linux version 5.4.233-qgki-16133-g886f0b784ccc (mesa@salvo-z390aorusmaster) (Android (6877366 based on r383902b1) clang version 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79), LLD 11.0.2 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79)) #1 SMP PREEMPT Sat Jan 13 14:58:48 CET 2024
> [+] kallsyms linux_banner offset: 0x22ff904
> [+] kernel version major: 5, minor: 4, patch: 233 
> [+] kallsyms kallsyms_token_table offset: 0x01f74780
> [+] kallsyms endian: little
> [+] kallsyms kallsyms_token_index offset: 0x01f74aa8
> [?] kallsyms can't find arm64 relocation table
> [+] kallsyms kallsyms_markers range: [0x01f73be8, 0x01f74780), count: 0x000002e6
> [+] kallsyms approximate kallsyms_offsets range: [0x01abc56c, 0x01b75bb4) count: 0x0002e592
> [+] kallsyms kallsyms_names offset: 0x01b75bc0
> [+] kallsyms kallsyms_num_syms offset: 0x01b75bb8, value: 0x0002e590
> [+] kallsyms names table linux_banner index: 0x0001b391
> [+] kallsyms sure linux_banner index: 0
> [+] kallsyms kallsyms_offsets offset: 0x01abc570
> [+] kallsyms tcp_init_sock: type: T, offset: 0x0146b490
> [+] kptools map_start: 0x146b490, max_size: 0x800
> [+] kallsyms kallsyms_lookup_name: type: T, offset: 0x00375970
> [+] kallsyms printk: type: T, offset: 0x0031ebcc
> [+] kallsyms paging_init: type: T, offset: 0x025eb524
> [+] kallsyms memblock_reserve: type: T, offset: 0x004a0d74
> [+] kallsyms memblock_phys_alloc_try_nid: type: T, offset: 0x026052d4
> [+] kallsyms memblock_mark_nomap: type: T, offset: 0x004a1028
> [+] kallsyms memstart_addr: type: D, offset: 0x02435480
> [+] kallsyms vabits_actual: type: D, offset: 0x024ed038
> [+] kallsyms kimage_voffset: type: D, offset: 0x024354a0
> [+] kptools supercall key: test
> [+] kallsyms panic: type: T, offset: 0x00290410
> [+] kallsyms rest_init: type: T, offset: 0x016fd2c4
> [+] kallsyms cgroup_init: type: T, offset: 0x025f89e8
> [?] kallsyms no symbol: kernel_init
> [?] kallsyms no symbol: report_cfi_failure
> [?] kallsyms no symbol: __cfi_slowpath_diag
> [+] kallsyms __cfi_slowpath: type: T, offset: 0x00419e20
> [+] kallsyms copy_process: type: t, offset: 0x0028d008
> [+] kallsyms cgroup_post_fork: type: T, offset: 0x003803d8
> [+] kallsyms __do_execve_file: type: t, offset: 0x004e1838
> [?] kallsyms no symbol: do_execveat_common
> [?] kallsyms no symbol: do_execve_common
> [+] kallsyms avc_denied: type: t, offset: 0x00716e24
> [+] kallsyms slow_avc_audit: type: T, offset: 0x00715ed8
> [+] kallsyms input_handle_event: type: t, offset: 0x00c853c4
> [+] kallsyms vfs_statx: type: T, offset: 0x004ddd8c
> [?] kallsyms no symbol: do_statx
> [?] kallsyms no symbol: vfs_fstatat
> [+] kallsyms do_faccessat: type: T, offset: 0x004d17d8
> [?] kallsyms no symbol: sys_faccessat
> [+] kptools patch done: Image_patched
> ```

Yes sir, We need to disable all the samsung anti root techniques like RKP, Defex and integrity subsystem to make APatch work :-). Not working in every samsung device btw.

Fede2782, 2024-01-14:
Could I patch using Magisk boot to disable security features and then patch with APatch?

BlackMesa123, 2024-01-15:
> Could I patch using Magisk boot to disable security features and then patch with APatch?

As explained in https://github.com/bmax121/APatch/issues/123#issuecomment-1890473160, those patches are old and don't work on newer devices (tho they're still there to support them), the non-booting issue seems also not related to either RKP_NS or DEFEX.

@bmax121 you might want to check `CONFIG_RKP` macros in Samsung kernel as after a few tests I'm pretty sure this is what breaks boot. The fact I couldn't get any logs in boot logs probably means the kernel patching itself is faulty. `CONFIG_RKP` macros are defined in the following files:
```console
❯ grep -r -w 'CONFIG_RKP'
arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
arch/arm64/include/asm/pgtable.h:#ifdef CONFIG_RKP
arch/arm64/include/asm/pgtable.h:#ifdef CONFIG_RKP
arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
arch/arm64/mm/mmu.c://#ifndef CONFIG_RKP
arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
arch/arm64/net/bpf_jit_comp.c:#ifdef CONFIG_RKP
arch/arm64/net/bpf_jit_comp.c:#ifdef CONFIG_RKP
drivers/uh/Makefile:obj-$(CONFIG_RKP)   += rkp.o
init/main.c:#ifdef CONFIG_RKP
init/main.c:#ifdef CONFIG_RKP
init/main.c:#ifdef CONFIG_RKP
mm/slub.c:#ifdef CONFIG_RKP
mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
mm/slub.c:#ifdef CONFIG_RKP
kernel/bpf/core.c:#ifdef CONFIG_RKP
kernel/bpf/core.c:#ifdef CONFIG_RKP
kernel/module.c:#ifdef CONFIG_RKP
kernel/module.c:#ifdef CONFIG_RKP
kernel/module.c:#ifdef CONFIG_RKP
kernel/module.c:#ifdef CONFIG_RKP
kernel/module.c:#ifdef CONFIG_RKP
kernel/module.c:#ifdef CONFIG_RKP
```

ravindu644, 2024-01-15:
> > Could I patch using Magisk boot to disable security features and then patch with APatch?
> 
> As explained in [#123 (comment)](https://github.com/bmax121/APatch/issues/123#issuecomment-1890473160), those patches are old and don't work on newer devices (tho they're still there to support them), the non-booting issue seems also not related to either RKP_NS or DEFEX.
> 
> @bmax121 you might want to check `CONFIG_RKP` macros in Samsung kernel as after a few tests I'm pretty sure this is what breaks boot. The fact I couldn't get any logs in boot logs probably means the kernel patching itself is faulty. `CONFIG_RKP` macros are defined in the following files:
> 
> ```
> ❯ grep -r -w 'CONFIG_RKP'
> arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
> arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
> arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
> arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
> arch/arm64/include/asm/pgtable.h:#ifdef CONFIG_RKP
> arch/arm64/include/asm/pgtable.h:#ifdef CONFIG_RKP
> arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> arch/arm64/mm/mmu.c://#ifndef CONFIG_RKP
> arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
> arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
> arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
> arch/arm64/net/bpf_jit_comp.c:#ifdef CONFIG_RKP
> arch/arm64/net/bpf_jit_comp.c:#ifdef CONFIG_RKP
> drivers/uh/Makefile:obj-$(CONFIG_RKP)   += rkp.o
> init/main.c:#ifdef CONFIG_RKP
> init/main.c:#ifdef CONFIG_RKP
> init/main.c:#ifdef CONFIG_RKP
> mm/slub.c:#ifdef CONFIG_RKP
> mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
> mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
> mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
> mm/slub.c:#ifdef CONFIG_RKP
> kernel/bpf/core.c:#ifdef CONFIG_RKP
> kernel/bpf/core.c:#ifdef CONFIG_RKP
> kernel/module.c:#ifdef CONFIG_RKP
> kernel/module.c:#ifdef CONFIG_RKP
> kernel/module.c:#ifdef CONFIG_RKP
> kernel/module.c:#ifdef CONFIG_RKP
> kernel/module.c:#ifdef CONFIG_RKP
> kernel/module.c:#ifdef CONFIG_RKP
> ```

[image]
[image]

ravindu644, 2024-01-15:
> As explained in [#123 (comment)](https://github.com/bmax121/APatch/issues/123#issuecomment-1890473160), those patches are old and don't work on newer devices (tho they're still there to support [...]

See this. (All samsung anti root mechanisms) - https://github.com/ravindu644/APatch/tree/main/docs/guides/kernel_compilation#how-to-disable-kernel-securities--enable-the-required-features-from-menuconfig

Fede2782, 2024-01-15:
> > > Could I patch using Magisk boot to disable security features and then patch with APatch?
> > 
> > 
> > As explained in [#123 (comment)](https://github.com/bmax121/APatch/issues/123#issuecomment-1890473160), those patches are old and don't work on newer devices (tho they're still there to support them), the non-booting issue seems also not related to either RKP_NS or DEFEX.
> > @bmax121 you might want to check `CONFIG_RKP` macros in Samsung kernel as after a few tests I'm pretty sure this is what breaks boot. The fact I couldn't get any logs in boot logs probably means the kernel patching itself is faulty. `CONFIG_RKP` macros are defined in the following files:
> > ```
> > ❯ grep -r -w 'CONFIG_RKP'
> > arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
> > arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
> > arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
> > arch/arm64/include/asm/pgalloc.h:#ifdef CONFIG_RKP
> > arch/arm64/include/asm/pgtable.h:#ifdef CONFIG_RKP
> > arch/arm64/include/asm/pgtable.h:#ifdef CONFIG_RKP
> > arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> > arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> > arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> > arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> > arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> > arch/arm64/mm/mmu.c:#ifdef CONFIG_RKP
> > arch/arm64/mm/mmu.c://#ifndef CONFIG_RKP
> > arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
> > arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
> > arch/arm64/mm/pgd.c:#ifdef CONFIG_RKP
> > arch/arm64/net/bpf_jit_comp.c:#ifdef CONFIG_RKP
> > arch/arm64/net/bpf_jit_comp.c:#ifdef CONFIG_RKP
> > drivers/uh/Makefile:obj-$(CONFIG_RKP)   += rkp.o
> > init/main.c:#ifdef CONFIG_RKP
> > init/main.c:#ifdef CONFIG_RKP
> > init/main.c:#ifdef CONFIG_RKP
> > mm/slub.c:#ifdef CONFIG_RKP
> > mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
> > mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
> > mm/slub.c:#if defined(CONFIG_KDP) && defined(CONFIG_RKP)
> > mm/slub.c:#ifdef CONFIG_RKP
> > kernel/bpf/core.c:#ifdef CONFIG_RKP
> > kernel/bpf/core.c:#ifdef CONFIG_RKP
> > kernel/module.c:#ifdef CONFIG_RKP
> > kernel/module.c:#ifdef CONFIG_RKP
> > kernel/module.c:#ifdef CONFIG_RKP
> > kernel/module.c:#ifdef CONFIG_RKP
> > kernel/module.c:#ifdef CONFIG_RKP
> > kernel/module.c:#ifdef CONFIG_RKP
> > ```
> 
> ![IMG_20240115_141632_281](https://private-user-images.githubusercontent.com/126038496/296700897-4beb2600-133c-435d-a200-6a8424de65b2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUzMTcyOTMsIm5iZiI6MTcwNTMxNjk5MywicGF0aCI6Ii8xMjYwMzg0OTYvMjk2NzAwODk3LTRiZWIyNjAwLTEzM2MtNDM1ZC1hMjAwLTZhODQyNGRlNjViMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExNVQxMTA5NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZWQxMGI1OWNkNDRkOWUxMTkwZDI2NDAyYjM4Y2RlMDNlMzkxNzA5ZjliNDkzM2MzOGI4ZmYxM2Y0YTFhOTMxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kEfZmRfx-zuxk3oT9N0b1KCqg7EoISStpp5A6spZxTQ) ![IMG_20240115_141637_516](https://private-user-images.githubusercontent.com/126038496/296700911-9e89e1aa-60d2-42f5-936e-db6773fd1d3b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUzMTcyOTMsIm5iZiI6MTcwNTMxNjk5MywicGF0aCI6Ii8xMjYwMzg0OTYvMjk2NzAwOTExLTllODllMWFhLTYwZDItNDJmNS05MzZlLWRiNjc3M2ZkMWQzYi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExNVQxMTA5NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZTk4YTU1YTY4ZjY2ZjM3NzI1NjE5MzUyNjc4NTVmOThlOTE2NTBkNmZmMTk4OGJlYTBmZjA0YjZmM2RkNWFhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xx7N86ZKXDBKT-c6wOVVCn6Tacoo-zXFAFLHTmxKbqQ)

That's useful but all Mediatek Samsung sources are broken so this way could not be done on Mtk

Fede2782, 2024-01-15:
Currently building kernel is not even a solution on Samsung Mediatek devices because their kernel source is uncomplete and building is impossible without loosing full connectivity or something else

BlackMesa123, 2024-01-15:
> Currently building kernel is not even a solution on Samsung Mediatek devices because their kernel source is uncomplete and building is impossible without loosing full connectivity or something else

Also, isn't this project's main goal to patch kernel images automatically? Building custom kernel images just for Samsung devices is redundant, might as well just stick with KSU.

Fede2782, 2024-01-15:
> Also, isn't this project's main goal to patch kernel images automatically? Building custom kernel images just for Samsung devices is redundant, might as well just stick with KSU.

Yes, yes, of course. I just said that building wasn't an option

Sahil12524, 2024-01-21:
so what now, should we wait or leave rooting the device?
My device: Samsung M127G
stuck on boot logo and after few minutes it reboots itself and loops over!

bmax121, 2024-02-15:
I still don’t understand the principles of RKP and Defex, I’ll figure it out when I have time.

BlackMesa123, 2024-02-15:
> I still don’t understand the principles of RKP and Defex, I’ll figure it out when I have time.

Defex isn't relevant to the issue, this is mainly caused by those security components running on microHypervisor (µH) such as RKP and KDP. What I believe is happening is those alter the kernel struct in some way, so KernelPatch generates a malformed kernel image in the end, explaining why there's no output in kmsg when trying to debug the issue.

BlackMesa123, 2024-02-15:
Regarding https://github.com/ravindu644/APatch-Samsung, I've already explained in https://github.com/bmax121/APatch/issues/123#issuecomment-1890473160 those patches are useless in this case because they don't address this issue. RKP namespace protection is not anymore a thing in modern device and it would've been visible in kmsg if this was the case (eg. https://github.com/topjohnwu/Magisk/pull/7665#issuecomment-1937081802), while Defex LSM only blocks userspace executables basing off its set of rules/policy. This one also automatically disables if an unlocked bootloader status is detected (eg. https://github.com/BlackMesa123/android_kernel_samsung_s5e8835/blob/ed39d840e85ab23495efb36001d0cd792862c5c6/security/samsung/defex_lsm/core/defex_lsm.c#L68-L77)

InfiniteVariations1, 2024-02-16:
I have personally tried patching my stock boot.img file with APatch and I am having the same problem. I am currently using the Galaxy Tab S9 FE. I honestly suprised as I can boot just fine with KSU GKI kernels just fine; the only problem I have with that I can't read or write anything on my SD Card. I will be sticking with KSU for now.

Fede2782, 2024-02-17:
> I have personally tried patching my stock boot.img file with APatch and I am having the same problem. I am currently using the Galaxy Tab S9 FE. I honestly suprised as I can boot just fine with KSU [...]

Samsung doesn't use standard GKI. Samsung has its GKI sources which are device specific but contain all needed drivers like SDCard or SPen hardware. This means that GKI Samsung devices may miss some small features when using non Samsung kernel

InfiniteVariations1, 2024-02-17:
@Fede2782 Thanks for the information! Yes, this is why I am excited for APatch. As I mentioned the SDCard isn't being read; APatch could fix that issue by patching the boot.img specific to my device. I am using KSU for now and it is less detectable than Magisk. I don't how to compile my own kernel and I am missing out on some small things (actually the SPen works completely fine), so that's why I am currently waiting for APatch to support Samsung devices.

InfiniteVariations1, 2024-02-18:
> Samsung doesn't use standard GKI. Samsung has its GKI sources which are device specific but contain all needed drivers like SDCard or SPen hardware. This means that GKI Samsung devices may miss some [...]

Is there anyway I can access those resources? I am thinking about building a custom kernel for my S9 FE. I can't seem to find any source code at the moment which makes sense as it was released back in October 2023.

BlackMesa123, 2024-02-18:
> Is there anyway I can access those resources? I am thinking about building a custom kernel for my S9 FE. I can't seem to find any source code at the moment which makes sense as it was released back in October 2023.

https://opensource.samsung.com/uploadSearch?searchValue=X516

Fede2782, 2024-02-18:
> Is there anyway I can access those resources? I am thinking about building a custom kernel for my S9 FE. I can't seem to find any source code at the moment which makes sense as it was released back [...]

There is just Android 13 kernel. I am sending every week a request to Samsung for P615 and X516B sources without any response

InfiniteVariations1, 2024-02-18:
> > Is there anyway I can access those resources? I am thinking about building a custom kernel for my S9 FE. I can't seem to find any source code at the moment which makes sense as it was released back in October 2023.
> 
> https://opensource.samsung.com/uploadSearch?searchValue=X516

Thanks! Unfortunately there is no source code for my device.

InfiniteVariations1, 2024-02-18:
> > Is there anyway I can access those resources? I am thinking about building a custom kernel for my S9 FE. I can't seem to find any source code at the moment which makes sense as it was released back in October 2023.
> 
> https://opensource.samsung.com/uploadSearch?searchValue=X516
Nevermind! I found the source code! Thanks man! Also big fan of your work! Especially KnoxPatch!

InfiniteVariations1, 2024-02-18:
> There is just Android 13 kernel. I am sending every week a request to Samsung for P615 and X516B sources without any response

Will hopefully you will get is soon. :)

bmax121, 2024-02-19:
I guess not in a short time, I still have to work for the company

InfiniteVariations1, 2024-02-19:
> I guess not in a short time, I still have to work for the company

Sorry for asking, but what exactly do you mean?

InfiniteVariations1, 2024-02-19:
> > Is there anyway I can access those resources? I am thinking about building a custom kernel for my S9 FE. I can't seem to find any source code at the moment which makes sense as it was released back in October 2023.
> 
> https://opensource.samsung.com/uploadSearch?searchValue=X516

Hey, I am new to building kernels, but does the source code have to match with the exact model and version? I have the SM-X510 with Android 14 on it. I currently have the source code for the specific model, but it is for Android 13. The current GKI kernel is Android-13-5.15.104.

privacyguy123, 2024-03-11:
Thread is unclear - is it safe to try and patch Samsung devices?

Glennza1962, 2024-03-13:
> Thread is unclear - is it safe to try and patch Samsung devices?

I just tried to install APatch on my S9 with custom ROM. Patch written and installed okay but just bootloops. So I'd say 'no'.

xxjoe2, 2024-04-07:
> ![IMG_20240115_141632_281](https://private-user-images.githubusercontent.com/126038496/296700897-4beb2600-133c-435d-a200-6a8424de65b2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY [...]

sorry a bit confused. that photo with a tablet... is that yours?  so you did it on the samsung tablet?

yoro1836, 2024-05-08:
I don't know if it will work, I have 5.4 gki boot img. If you want, I can share it. (Not Patched)

CCCC-L, 2024-07-14:
On Samsung TabS7+ (T970) it works fine with `KernelPatch 0.10.7` but not with `KernelPatch 0.11.0-dev` Perhaps more defconfig needs to be disabled.

xxjoe2, 2024-07-14:
> On Samsung TabS7+ (T970) it works fine with `KernelPatch 0.10.7` but not with `KernelPatch 0.11.0-dev` Perhaps more defconfig needs to be disabled.

stock or modded kernel?

CCCC-L, 2024-07-14:
> > On Samsung TabS7+ (T970) it works fine with `KernelPatch 0.10.7` but not with `KernelPatch 0.11.0-dev` Perhaps more defconfig needs to be disabled.
> 
> stock or modded kernel?

modded kernel

https://github.com/CCCC-L/kernel_T970/commit/eda04fe06516eee968cf9b684f73e43dcdecf84a
https://github.com/CCCC-L/kernel_T970/commit/c27ba36de3afdafa389de76d2e1b5a5aba7623f6

xxjoe2, 2024-07-14:
ok. i asked because i had similar problems before. there were too many changes from the dev and he have disabled some logs, eg the debug kernel was turm off in the code but not in the config file. so i have to trace the changes and rolled them back for enabling the debug kernel config

just my 2 cents and i hope this helps

good luck

EslamGamalEssa, 2024-08-16:
hello when this will be solved for the a52s 5g (a528b) device please if anyone can answer it would be very appreciated

thanks

EslamGamalEssa, 2024-08-17:
please add support for samsung devices

EslamGamalEssa, 2024-08-23:
> Owner
please sir fix this issue

EslamGamalEssa, 2024-08-23:
> I still don’t understand the principles of RKP and Defex, I’ll figure it out when I have time.

please sir fix this issue

EslamGamalEssa, 2024-09-04:
@bmax121 in wich century you will solve this issue?
it's already 2050 happy new year BTW
please take a little bit of your time to look for this

closed 2024-09-04 by bmax121

EslamGamalEssa, 2024-09-04:
@bmax121 you can just go to hell buddy

little dogs like you should know better

tuyie78, 2024-09-04:
> @bmax121 you can just go to hell buddy
> 
> little dogs like you should know better

lol

madodream89, 2024-10-12:
hello i am trying installing apatch in note n9600 but is semicompatible and magisk modules not work i try uninstalling magisk give bootloops and reboots. the problem is the multidiseable and kribit kernel, the source code has magisk code

i need a version of multidiseable and kribit kernel without magisk boot patch and code

https://github.com/corsicanu/multidisabler-samsung
[update-binary.zip](https://github.com/user-attachments/files/17350728/update-binary.zip)
[update-binary.zip](https://github.com/user-attachments/files/17350738/update-binary.zip)

https://github.com/klabit87/android_kernel_samsung_sdm845/tree/q

[update-binary.zip](https://github.com/user-attachments/files/17350732/update-binary.zip)
[update-binary.zip](https://github.com/user-attachments/files/17350737/update-binary.zip)

the open source file is here before samsung opensource deleted the file and i trying port postmarketos in n9600 https://gitlab.com/madodream89/samsung-galaxy-note-9-n9600-sdm845-postmarketos-port[](url)

madodream89, 2024-10-12:
https://gist.github.com/TheAirBlow/06bb6f49b4af9dd718d9321fbb8605b3

madodream89, 2024-10-12:
https://github.com/ianmacd/multidisabler-samsung/issues/14

MKindy, 2025-02-19:
> [@bmax121](https://github.com/bmax121) you can just go to hell buddy
> 
> little dogs like you should know better

Edgelord script kiddie. 😆

GoRhanHee, 2025-08-15:
> [image] [image]

bro, cant u fork the APatch repo and put the rkp and cfp removal function code in it and make it work on all Scamsung devices?