bmax121/APatch #125: oneplus ColorOS13.0，超级用户列表一直转圈圈

ttt789-ppo, 2024-01-13:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

超级用户一直转圈圈，刷新不出应用，我看了系统的应用行为记录，它是被允许读取应用列表的，但是它就是超级用户那里转圈圈
[image]
![Screenshot_2024-01-14-00-18-23-34_e6871cd13ad370b30dca80a8c4050ac0](https://github.com/bmax121/APatch/assets/72877282/b969cc87-f6df-42c7-81f9-1d1be8e838f8）

### To Reproduce

1.打开apatch
2.点击超级用户
3.可以看到一直转圈圈，显示不出应用

### Expected behavior

超级用户那里可以看到应用

### Screenshots

[image]
[image]

### Device info

- Device:OnePlus 11R
- OS Version:ColorOs13.0
- APatch Version:10419
- Kernel Version:0.8.3
[image]

bugme2, 2024-01-14:
我这边14给了读取应用列表权限可以，但是13不会主动获取读取应用列表权限，所以一直转圈圈

ttt789-ppo, 2024-01-14:
> 我这边14给了读取应用列表权限可以，但是13不会主动获取读取应用列表权限，所以一直转圈圈

必须升级14吗？可是我看ksu也是一样不用权限但能读取应用列表，c14系统不好用才在13的

857261741, 2024-01-16:
> > 我这边14给了读取应用列表权限可以，但是13不会主动获取读取应用列表权限，所以一直转圈圈
> 
> 必须升级14吗？可是我看ksu也是一样不用权限但能读取应用列表，c14系统不好用才在13的

不用升级，只需要给读取应用列表的权限就可以了，需要自己去软件后台给，这个没有请求权限弹窗

29846, 2024-01-16:
[image]
奇了个怪，不能手动授予他权限

bugme2, 2024-01-17:
> ![截图_2024-01-16-20-02-26-78_fc704e6b13c4fb26bf5e411f75da84f2](https://private-user-images.githubusercontent.com/156798179/297039895-2d99fd38-d5c3-49b7-8ea6-4ad6979a3f5b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzEyNTAsIm5iZiI6MTcwNTQ3MDk1MCwicGF0aCI6Ii8xNTY3OTgxNzkvMjk3MDM5ODk1LTJkOTlmZDM4LWQ1YzMtNDliNy04ZWE2LTRhZDY5NzlhM2Y1Yi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExN1QwNTU1NTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yOThmMGE0ODY1NWVjYzQzY2VkZjJjNTE0NTA3ZjJkZTUzZWI5Y2I3YjI4ODY0NThjNDdhMWUzMzY5YmM3NTY1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.HUxhsdB3pZvXqbgVAxUdp1JpxazKA07xZD5uZBV6FQQ) 奇了个怪，不能手动授予他权限

13是不能手动的，需要软件主动申请过后才能手动，14软件不申请也能手动给

ttt789-ppo, 2024-01-17:
> ![Screenshot_2024-01-16-20-02-26-78_fc704e6b13c4fb26bf5e411f75da84f2](https://private-user-images.githubusercontent.com/156798179/297039895-2d99fd38-d5c3-49b7-8ea6-4ad6979a3f5b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzI0NTksIm5iZiI6MTcwNTQ3MjE1OSwicGF0aCI6Ii8xNTY3OTgxNzkvMjk3MDM5ODk1LTJkOTlmZDM4LWQ1YzMtNDliNy04ZWE2LTRhZDY5NzlhM2Y1Yi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExN1QwNjE1NTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZmU1ZTE5ZmQzODczYTYyN2M5YzE3Mjc4ZTkzMjdlN2JiYmQzNmI1ODRlMTFmNDAzYjdiZjQ0MjM2MDg2MGQ0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.GEqsxsJ5upfqKBYstjULqWdmVjRjWuChIXpgv3a1p70) 奇了个怪，不能手动授予他权限

安装c14的权限管理就行，找c14手机提取一个安装包

ttt789-ppo, 2024-01-17:
> 13是不能手动的，需要软件主动申请过后才能手动，14软件不申请也能手动给

c13可以安装c14手机提取的权限管理，安装上就能手动授权了

bmax121, 2024-01-17:
fixed

closed 2024-01-17 by bmax121