bmax121/APatch #129: Why was my question closed?  I will submit it only if there must be a problem.

suman666, 2024-01-14:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

After installing the patched image, my system wouldn't start, and the phone got stuck at the oneplus logo (not bootlooping).

### To Reproduce

The problem is consistent

### Expected behavior

Request the author to adapt the kernel!  I can provide boot.img

### Device info

- Device:OnePlus 9 Pro
- OS Version:ColorOS 11.2（Android 11）
- APatch Version:10420
- Kernel Version:5.4.61-qgki-ga139308

Yervant7, 2024-01-14:
This is not the right place for the issue, please go to the KernelPatch repository

suman666, 2024-01-14:
> 这不是问题的正确位置，请转到 KernelPatch 存储库

Thanks, I've submitted it in KernelPatch.

suman666, 2024-01-14:
Sorry, I was too stupid.

closed 2024-01-14 by suman666