bmax121/APatch #130: Q1.how to use Apatch App Shell cmd Code:  cp   /data/app/apatch.apk to /system/app/apatch.apk  in android 8+;Q2. how to phone power on Auto Run Apatch

pigadmin, 2024-01-14:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

such as title

### Device info

- Device:
- OS Version:
- APatch Version:
- Kernel Version:

pigadmin, 2024-01-14:
Magisk app after  reboot recovery achieve Q.1and Q.2

marat2509, 2024-01-14:
Describe your problem in more detail, with examples.

Zenlua, 2024-01-14:
Maybe the su command has a problem, try typing the command: `pm` it will show an error, cmd commands don't work because su has a problem.

bmax121, 2024-01-17:
/system/app is read only, you can't copy file to it

closed 2024-01-17 by bmax121

pigadmin, 2024-01-19:
> /system/app is read only, you can't copy file to it

# stub.apk           binary    The stub Magisk app to embed into ramdisk.
./magiskboot compress=xz stub.apk stub.xz

provide some methods to  pack apk to xz file  if it's alright.
after android os recovery finish  resume apk file or other files.

[image]

[image]

[image]