bmax121/APatch #132: Starting the APatch system will restart

ziki11, 2024-01-14:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

Flash the patched boot，after restarting the phone，starting the APatch system will restart.

### Device info

- Device:Redmi note 11 5G
- OS Version:Android 13
- APatch Version:10420
- Kernel Version:4.14.186

Yervant7, 2024-01-14:
Logs?

ziki11, 2024-01-14:
Flash back to the original boot to enable APatch.
[APatch_bugreport_2024-01-15_01_29.tar_012944.gz](https://github.com/bmax121/APatch/files/13932515/APatch_bugreport_2024-01-15_01_29.tar_012944.gz)

Yervant7, 2024-01-15:
Did you mean that the system is in bootloop? If yes then these logs are useless try changing Kernel

ziki11, 2024-01-15:
No, after flashing the patched boot.img, the system can boot up normally and other software can be used normally. However, when I click on APatch, the system will restart. This may be a ROM issue, and I will try to change the boot.

bmax121, 2024-01-17:
Does it crash when just open APatch or click some button inside APatch app

Jehao, 2024-01-18:
Redmi Note 11 Pro代号pissarro也是如此，在刷完apatch修补的boot后重启再进入apatch后输入密钥
[image]
会突然屏幕一黑并重启

pomelohan, 2024-03-05:
> Redmi Note 11 Pro代号pissarro也是如此，在刷完apatch修补的boot后重启再进入apatch后输入密钥 ![25A7543EDFE1FC7FC8800D4E7F587941](https://private-user-images.githubusercontent.com/118976745/297675352-c5f9979e-276c-4ac3-9f56-4795fd1860ca.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk2NDkwODcsIm5iZiI6MTcwOTY0ODc4NywicGF0aCI6Ii8xMTg5NzY3NDUvMjk3Njc1MzUyLWM1Zjk5NzllLTI3NmMtNGFjMy05ZjU2LTQ3OTVmZDE4NjBjYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNVQxNDI2MjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YmZlODRkZjhlMTdhNDEyMjRmM2ZhZmY4MWUxYjI4ZGMwN2NkYzFlMGIwODAxYWM5NDIyNmFjZjhhODVjYjk4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.9GtCwVjI9LSYZEa0t7cV9OcPqek-OUpeck0CsgbFeIc) 会突然屏幕一黑并重启

It's a known issue that kernel panicked but not manager FC. Wait for fixing in future.

pomelohan, 2024-03-05:
> Flash back to the original boot to enable APatch. [APatch_bugreport_2024-01-15_01_29.tar_012944.gz](https://github.com/bmax121/APatch/files/13932515/APatch_bugreport_2024-01-15_01_29.tar_012944.gz)

Android System crash's log should be caught by "adb logcat" before you open APatch app and then trigger crash. From APatch's Bugreport couldn't get enough log. But I found the possible problem from your log, it seems that it's your ROM-side's problem that `seccomp prevented call to disallowed arm64 system call`, I thought it didn't belong to APatch's bug. Ask your ROM maintainer instead.

`F libc    : Fatal signal 31 (SIGSYS), code 1 (SYS_SECCOMP) in tid 6640 (dmesg), pid 6640 (dmesg)
I crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstoneProto
I crash_dump64: performing dump of process 6640 (target tid = 6640)
F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
F DEBUG   : Build fingerprint: 'Redmi/evergo_cn/evergo:13/TP1A.220624.014/V14.0.3.0.TGBCNXM:user/release-keys'
F DEBUG   : Revision: '0'
F DEBUG   : ABI: 'arm64'
F DEBUG   : Timestamp: 2024-01-15 01:28:35.476367571+0800
F DEBUG   : Process uptime: 1s
F DEBUG   : Cmdline: dmesg
F DEBUG   : pid: 6640, tid: 6640, name: dmesg  >>> dmesg <<<
F DEBUG   : uid: 10507
F DEBUG   : signal 31 (SIGSYS), code 1 (SYS_SECCOMP), fault addr --------
F DEBUG   : Cause: seccomp prevented call to disallowed arm64 system call 116
F DEBUG   :     x0  000000000000000a  x1  0000000000000000  x2  0000000000000000  x3  0000000000000000
F DEBUG   :     x4  0000007fe58763c1  x5  0000000000000004  x6  0000803e0d070a4e  x7  2e6d686171ff372c
F DEBUG   :     x8  0000000000000074  x9  69d918b2f6cec900  x10 0000007fe5874320  x11 0000007fe58742f0
F DEBUG   :     x12 ffffff80ffffffd0  x13 0000000000000040  x14 0000000000000040  x15 0000000000000000
F DEBUG   :     x16 0000005e4e43b930  x17 0000007f4a0d4c50  x18 0000007f50e50000  x19 0000000000000000
F DEBUG   :     x20 0000000000000000  x21 0000007f50814000  x22 0000005e4e43f408  x23 0000005e4e440458
F DEBUG   :     x24 0000005e4e43caf0  x25 0000000000000000  x26 0000000000000000  x27 0000000000000000
F DEBUG   :     x28 0000000000000000  x29 0000007fe58763d0
F DEBUG   :     lr  0000005e4e419668  sp  0000007fe58743c0  pc  0000007f4a0d4c58  pst 0000000060001000
F DEBUG   : backtrace:
F DEBUG   :       #00 pc 00000000000b5c58  /apex/com.android.runtime/lib64/bionic/libc.so (klogctl+8) (BuildId: 79a04478da0708803eb615bfb54e2a39)
F DEBUG   :       #01 pc 0000000000053664  /system/bin/toybox (dmesg_main+436) (BuildId: a4c48872855ca9010b19aadb5f925130)
F DEBUG   :       #02 pc 0000000000035188  /system/bin/toybox (toy_exec_which+88) (BuildId: a4c48872855ca9010b19aadb5f925130)
F DEBUG   :       #03 pc 0000000000034ab8  /system/bin/toybox (toybox_main+132) (BuildId: a4c48872855ca9010b19aadb5f925130)
F DEBUG   :       #04 pc 0000000000035200  /system/bin/toybox (main+112) (BuildId: a4c48872855ca9010b19aadb5f925130)
F DEBUG   :       #05 pc 000000000004b8c0  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+96) (BuildId: 79a04478da0708803eb615bfb54e2a39)`

Close it since it's not APatch's bug. Feel free to re-open the issue if you provide a better log or still have issues.

closed 2024-03-05 by pomelohan