bmax121/APatch #135: [Zygisk Next] monitor:❌stopped(zygote crashed)

cherinyy, 2024-01-15:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

[Zygisk Next](https://github.com/Yervant7/ZygiskNext) is not working.

### Logs

[APatch_bugreport_2024-01-15_13_59.tar.gz](https://github.com/bmax121/APatch/files/13935425/APatch_bugreport_2024-01-15_13_59.tar.gz)

### Device info

- Device: OnePlus 3T
- OS Version: lineage-18.1-20240105-nightly-oneplus3
- APatch Version: 10420-8-g816d628
- Kernel Version: 3.18.124-lineageos-g04ff888001f2

### Additional context

Device Info: https://wiki.lineageos.org/devices/oneplus3/variant2/

cc @Yervant7

Yervant7, 2024-01-15:
Do you have another root solution installed?

Yervant7, 2024-01-15:
I opened the issues in my repository let's go there

cherinyy, 2024-01-15:
https://github.com/Yervant7/ZygiskNext/issues/4

closed 2024-01-15 by cherinyy