bmax121/APatch #138: Apps are adding itself without the user permissions.

ravindu644, 2024-01-15:
This weird issue happened after the lastest update, and it's not a problem only with me.

[image]

[image]

sinoobie, 2024-01-16:
delete app package name that not you want to automatically enable super user permission after reboot on /data/adb/ap/package_config file then reboot your phone.

ravindu644, 2024-01-16:
> delete app package name that not you want to automatically enable super user permission after reboot on /data/adb/ap/package_config file then reboot your phone.

[image]
🙁

bmax121, 2024-01-17:
com.byyoung.setting has root permission, and it's uid is the same with by me a coffee (10381). It's odd. so upload the logs which is guided in issue template

symbuzzer, 2024-01-30:
Same issue here :(

bmax121, 2024-03-08:
This issue occurs when you uninstall or install a new app. Try the latest version, closing permissions should prevent it from happening. If it still doesn't work, reopen the issue.

closed 2024-03-08 by bmax121