bmax121/APatch #144: 建议实现仅刷入Boot后开机出现无图标APP功能

Yue-love, 2024-01-16:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

正如标题所说
这里将一个无法安装第三方APP的设备 仅刷入Apatch修补的BOOT 开机后设备无法安装Apatch软件 导致无法使用ROOT
Magisk会在开机时生成一个无图标的APP 这时可直接覆盖安装同版本APK 然后打开Magisk软件管理ROOT

### To Reproduce

这里将一个无法安装第三方APP的设备 仅刷入Apatch修补的BOOT 开机后设备无法安装Apatch软件 导致无法使用ROOT

### Expected behavior

预期结果：
修补的BOOT刷入设备后 设备开机 出现无图标的Apatch-APP或直接自动安装Apatch-APP 

### Device info

- Device: EEBBK-S5 Pro
- OS Version: Study OS V6.3.0
- APatch Version: 10433
- Kernel Version: 4.4.153-perf

Yue-love, 2024-01-17:
这对一些设备非常重要，例如在一些学习机，默认无法安装第三方软件，如果可以实现这个功能，就可以覆盖更新

 
可以做，但没必要

bmax121, 2024-01-17:
或许可以多放个脚本在里面，安装后执行这个脚本，这样应该会更好

bmax121, 2024-01-29:
#173

closed 2024-01-29 by bmax121