bmax121/APatch #152: juicessh cannot use su command

bugme2, 2024-01-17:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English, no reference to outside forums
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

[image]

### Device info

- Device: RMX1851
- OS Version: Android 11(RealmeUI 2.0)
- APatch Version: 10433
- Kernel Version: 0.8.4

bugme2, 2024-01-17:
还有MyAndroidTools也无法获取权限
[image]

bmax121, 2024-01-17:
upload logs in /data/adb/ap/log

bugme2, 2024-01-17:
> upload logs in /data/adb/ap/log

[log.zip](https://github.com/bmax121/APatch/files/13965528/log.zip)

ponces, 2024-01-21:
I have easily reproduced this issue and can confirm it.

I can also confirm that this is due to the fact that `su` is in fact being called using `/data/adb/apd` instead of `/system/bin/kp`.
Because of this, the `arg[0]` is not the one expected and that's why the root shell is not being invoked based on [this line](https://github.com/bmax121/APatch/blob/86e769850e4931ceffb312152a5403cb171c72f8/apd/src/cli.rs#L80).
@bmax121 should we consider `/data/adb/apd` on this `if` condition too or should we look into this in a different way?

ponces, 2024-01-24:
Fixed with #8d49ef3733b83d2aa6ef3390305eb996fb68c8fe.
Closed.

closed 2024-01-24 by ponces