bmax121/APatch #153: Unable to modify environment variables after using APatch

cherinyy, 2024-01-17:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English, no reference to outside forums
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

I want to configure the global environment variable `captive_portal_https_url` because Android tests HTTPS connectivity through `www.google.com` by default, which cannot be accessed in the Chinese mainland.

- [config.xml - Android Code Search](https://cs.android.com/android/platform/superproject/+/main:packages/modules/NetworkStack/res/values/config.xml;l=24 "config.xml - Android Code Search")
- [NetworkStackUtils.java - Android Code Search](https://cs.android.com/android/platform/superproject/+/main:packages/modules/NetworkStack/src/com/android/networkstack/util/NetworkStackUtils.java;l=152 "NetworkStackUtils.java - Android Code Search")

This won't be an issue on Android 12 and above, but on Android 7 - 11, if HTTPS access fails, the network is considered unavailable.

So I have to manually configure a server that can be accessed normally, such as `connectivitycheck.gstatic.com`.

```
adb shell settings put global captive_portal_https_url https://connectivitycheck.gstatic.com/generate_204
```

The problem is that I found that as soon as I restarted the device, the configuration became invalid.

### Logs

[APatch_bugreport_2024-01-15_13_59.tar.gz](https://github.com/bmax121/APatch/files/13958744/APatch_bugreport_2024-01-15_13_59.tar.gz)

### Device info

- Device: OnePlus 3T
- OS Version: lineage-18.1-20240105-nightly-oneplus3
- APatch Version: 10420-8-g816d628
- Kernel Version: 3.18.124-lineageos-g04ff888001f2

bmax121, 2024-01-17:
have you tested get captive_portal_https_url

cherinyy, 2024-01-17:
> have you tested get captive_portal_https_url

Yes, I also checked via [SetEdit](https://play.google.com/store/apps/details?id=by4a.setedit22) and after restarting `captive_portal_https_url` became `null`.

bmax121, 2024-03-07:
Maybe it's not the issue of APatch. Try magisk resetprop

closed 2024-03-07 by cherinyy