bmax121/APatch #154: after flash APatch boot. bluetooth no sound.

zerofo, 2024-01-17:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English, no reference to outside forums
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

- Bluetooth connect fine. 
- but no sound with the Bluetooth headset.

### To Reproduce

1. flash back the origin boot.img, can fix it.

2. flash the apatch one it can be reproduce

### Device info

- Device: xiaomi 12s ultra.
- OS Version:  Hyperos  24.1.8.DEV
- APatch Version: 10428
- Kernel Version: 5.10.177-android12-9-00001-g219d8dfbba07-ab10551810

### Additional context

[APatch_bugreport_2024-01-17_17_58.tar.gz](https://github.com/bmax121/APatch/files/13961488/APatch_bugreport_2024-01-17_17_58.tar.gz)

zerofo, 2024-01-17:
> That's difficult to figure out why. Does su works?

yes. it work fine.

zerofo, 2024-01-17:
> That's difficult to figure out why. Does su works?
Additional information:
1. flash the APatch's IMG in bootloader ( if flash it in the TWRP, sometime has some "mount error" about Bluetooth.  )
2. reboot.
then the Bluetooth headset and no sound.

it dont need to install the AndroidPatch. ofc install it, su is work.

zerofo, 2024-01-18:
It's probably kernel patch's problem.
`If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch`.
but it can boot.

zerofo, 2024-01-18:
10446 still no sound

ravindu644, 2024-01-18:
> 10446 still no sound

https://t.me/APatchChannel/42

zerofo, 2024-01-23:
it fix on 10472.

closed 2024-01-23 by zerofo