bmax121/APatch #163: Resetprop is outdated

Pat-JK, 2024-01-18:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English, no reference to outside forums
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

The latest version of the [Play Integrity Fork](https://github.com/osm0sis/PlayIntegrityFork) module by osm0sis fails to work as resetprop is missing the -Z argument.

### To Reproduce

1: Open a terminal as root
2: Attempt to use resetprop with -Z argument, (example: resetprop -Z ro.boot.flash.locked 1)
3: Will display help information instead of running the proper command

### Expected behavior

Command should run as expected

### Device info

- Device: Motorola One 5g Ace
- OS Version: 13
- APatch Version: 10450
- Kernel Version: 4.19.157

osm0sis, 2024-01-18:
`resetprop -Z` was added to Magisk about 8 months ago, and the latest Magisk stable was in late November.

KernelSU updated to latest resetprop from Magisk Canary 26404 about 2 weeks ago: https://github.com/tiann/KernelSU/commit/153ce9a39a25d1e3556d2711e92026d36b9405cf

Looks like resetprop was just missed for APatch when you did your binary updates to 26404 2 weeks ago as well: https://github.com/bmax121/APatch/tree/main/app/libs/arm64-v8a

sekaiacg, 2024-01-20:
Example from ksu: https://github.com/tiann/KernelSU/commit/153ce9a39a25d1e3556d2711e92026d36b9405cf

closed 2024-01-22 by bmax121