bmax121/APatch #166: Flashable apps don work as expected

Shubhamkr4, 2024-01-19:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English, no reference to outside forums
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

After flashing app modules, apps don't work as expected even after permitting su.
I flashed a dialer app and whenever I called someone it opened the g dialer even after disabling it.
This worked with magisk if that's needed

### Expected behavior

After calling someone it should open the dialer's interface and not g dialer's

### Screenshots

https://github.com/bmax121/APatch/assets/131890976/66839730-36bd-494c-87ed-5a55ab5f6c62

### Device info

- Device:berlin
- OS Version:13
- APatch Version:
- Kernel Version:5.4.233

Yervant7, 2024-01-20:
Disable exclude of the app in apatch su list

Shubhamkr4, 2024-01-30:
> Disable exclude of the app in apatch su list

Already tried before posting this issue.

Shubhamkr4, 2024-01-30:
Miui core and gallery also not working even after enabling su.
Also flashing miui core breaks browsers

Shubhamkr4, 2024-03-05:
Solved after 8.0.5

closed 2024-03-05 by Shubhamkr4