bmax121/APatch #171: Lsposed seems to be not supported

mmwong1984, 2024-01-20:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English, no reference to outside forums
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

Hmm..... I can't use lsposed, anyone have idea? I have tried with zygisknext

### Device info

- Device:
- OS Version:
- APatch Version:
- Kernel Version:

et-ness, 2024-01-20:
Have you tried [this](https://github.com/Yervant7/ZygiskNext) version of Zygisk Next?

RafaeloxMC, 2024-01-20:
You need to use the Zygisk fork for Apatch, someone else already sent a link in here

alex193a, 2024-01-20:
LSPosed is supported, but you must use the ZygiskNext fork by @Yervant7.
For this, use the link provided by @et-ness in the previous comment 😉

closed 2024-01-20 by alex193a