bmax121/APatch #174: 只能安装第一个模块，后续模块无法安装  /  Only the first module can be installed, subsequent modules cannot be installed

dzlea, 2024-01-21:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English, no reference to outside forums
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

安装第二个模块后显示: No such file or directory (os error 2)

After installing the second module: No such file or directory (os error 2)

### To Reproduce

安装第一个模块后，返回，再安装第二个模块显示报错（No such file or directory (os error 2)）

I can only install the first module, and then I get an error if I install a later module.

### Expected behavior

https://github.com/bmax121/APatch/assets/54771820/00e26fdd-41b1-493c-8478-e9f90dee451f

### Screenshots

[image]

[image]

### Logs

[APatch_bugreport_2024-01-21_13_11.tar.gz](https://github.com/bmax121/APatch/files/14000231/APatch_bugreport_2024-01-21_13_11.tar.gz)

### Device info

- Device:  oneplus 6t
- OS Version:  OxygenOS 11.1.2.2
- APatch Version: Latest
- Kernel Version: 4.9.227-perf+

dzlea, 2024-02-02:
> 一加6遇到同样的问题，请问解决了吗？ I'm experiencing the same problem with the Oneplus 6, has it been resolved yet?

没有    No

star1226, 2024-02-10:
试试在主页--系统补丁--卸载--安装，然后到安装模块那里安装新的模块

ys1231, 2024-02-18:
I encountered an update that was displayed.,After the download is successful.,There is no automatic installation.,Check the source code and find that the download is completed listening to the comment.,I don't know why the comment is out.
我遇到显示有更新,下载成功之后,没有自动安装,查看源码发现下载完成监听注释了,不知道是因为什么注释掉了.    
`// DownloadListener(context, onInstallModule)`

pomelohan, 2024-03-17:
Try latest version

pomelohan, 2024-04-01:
长时间未回复，暂时关闭，如有问题可重新开启

closed 2024-04-01 by pomelohan

zuiailaoda, 2024-04-23:
> 长时间未回复，暂时关闭，如有问题可重新开启

最新版本还是不行哦

pomelohan, 2024-04-23:
你确认你的版本是最新的 CI 版本吗

cakn233, 2024-04-28:
> 你确认你的版本是最新的 CI 版本吗

我有遇到同样的问题，用最新的cl版本解决了，正常刷入