bmax121/APatch #190: AndroidPatch doesn't install

cosbot1, 2024-01-30:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English. 请使用英文，不然会被关闭。
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

App crashes after clicking Install android patch. Kernel patch installed without problems

### To Reproduce

1. Install KernelPatch with heimdall
2. Click install on AndroidPatch

### Expected behavior

To install the AndroidPatch

### Screenshots

https://github.com/bmax121/APatch/assets/155786925/2c692d25-8fcd-48a9-aef1-c275e06229ee

### Logs

[APatch_bugreport_2024-01-30_01_57.tar.gz](https://github.com/bmax121/APatch/files/14091203/APatch_bugreport_2024-01-30_01_57.tar.gz)

### Device info

- Device: Galaxy s10e SM-G970F
- OS Version: LineageOS 21, Android 14
- APatch Version: 10446-26-gcbdda9a
- Kernel Version: 4.14.336-g2d66080c4b9c
- KernelPatch Version: 0.8.5

CASTOK544, 2024-02-26:
我是直接闪退

bmax121, 2024-03-08:
Try the latest version,  If it still doesn't work, reopen the issue.

closed 2024-03-08 by bmax121