bmax121/APatch #192: Running pm from Termux fails

larena1, 2024-01-30:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English. 请使用英文，不然会被关闭。
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

Running pm from Termux fails. This was never an issue with Magisk and KernelSU. Maybe SELinux issue?

$ sudo pm
cmd: Failure calling service package: Failed transaction (2147483646)

### To Reproduce

Open Termux
Run pm or sudo pm

### Expected behavior

Should not yield an error 

### Device info

- Device:
- OS Version: 13
- APatch Version:  10446
- Kernel Version: 4.9
- KernelPatch Version: 0.8.5

closed 2024-03-07 by bmax121

MrLumin06, 2024-03-08:
After the latest update, the problem is still occurring

APatch: 10952/KPatch: 0.10.2
Kernel: 4.19.157-perf-g8a3ae9055596

larena1, 2024-03-18:
It'd be great if you could open a new issue as I unfortunately can not reopen this. I'm sure @bmax121 just overlooked it and will fix it asap then

Typhonling, 2024-03-18:
That's okay, maybe he'll see responses to closed questions in his email. I think we should be patient.

larena1, 2024-03-18:
He gets lots of emails probably and can't read them all but when he checks the issues here, he'll probably only go over the open ones.