bmax121/APatch #199: Not working on Samsung phones

rajarshikhatua100, 2024-02-03:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English. 请使用英文，不然会被关闭。
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

After flashing the Apatch -ed boot image file the device is not booting up

### Device info

- Device: Samsung Galaxy M31
- OS Version:13
- APatch Version:10474
- Kernel Version:4.14
- KernelPatch Version:10474

RafaeloxMC, 2024-02-07:
According to Apatch telegram bot:
"apatch_samsung:
Why APatch gives bootloops in Samsung devices:
[](https://g.co/bard/share/d0502ae5b7e5)
*Fixable by disabling it in the kernel side"

Fede2782, 2024-02-10:
Duplicate of https://github.com/bmax121/APatch/issues/123.

By the way APatch doesn't work because of Knox DEFEX/RKP enabled in kernel which goes against rooting.

RafaeloxMC, 2024-02-10:
@Fede2782 i did post a link from the telegram bot, got removed tho :o

closed 2024-02-17 by rajarshikhatua100