bmax121/APatch #201: Motorola G51 5G not boot

Mxiier, 2024-02-05:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English. 请使用英文，不然会被关闭。
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

[image]
[image]
[image]
[image]
[image]

### Device info

- Device: moto G51
- OS Version: android 11
- APatch Version:
- Kernel Version: 5.4.86
- KernelPatch Version:

### Additional context

moto g51 not boot
[image]
[image]
[image]
[image]
[image]

bmax121, 2024-03-08:
Try the latest version, If it still doesn't work, wait for next update. In fact, if I don't have a phone, I can't solve it either.

closed 2024-03-08 by bmax121

Hellohi3654, 2024-10-23:
I have the same issue on the Motorola Edge 40
I tried the latest stable and latest actions build