bmax121/APatch #203: Case sensitive search bug

TheSingular, 2024-02-12:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English. 请使用英文，不然会被关闭。
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

When searching for installed apps to modify su permissions, the search is case insensitive when there are no spaces involved, on the other hand, if a space is inserted into the search bar, search becomes case sensitive.

### To Reproduce

1. Install any app with spaces in its name. (Or use an already installed app, "Kernel Flasher" in this example)
2. Open APatch
3. Go to Superuser tab
4. Search for the app without matching the case(like "kernel" or "KERNEL" or whatever)
5. Observe that the app disappears from the list as soon as a space is inserted (like "kernel ")
6. Go back and correct the case to exactly match the app (like "Kernel ")
7. Observe that the app returns to the list

### Expected behavior

App stays in the list when searching without exact case matching.

### Screenshots

[image]
[image]
[image]

### Logs

[APatch_bugreport_2024-02-12_09_07.tar.gz](https://github.com/bmax121/APatch/files/14235587/APatch_bugreport_2024-02-12_09_07.tar.gz)

### Device info

- Device: Xiaomi 12T
- OS Version: 14
- APatch Version:10472-2-gb56bd4b (10474)
- Kernel Version: 5.10.168-android12-9-.....
- KernelPatch Version: 0.8.5

pomelohan, 2024-03-06:
@TheSingular I had submitted a pull request and mentioned this issue. Can you try that build and tell me whether it is fixed?

TheSingular, 2024-03-06:
> @TheSingular Can you try this build and tell me whether it is fixed?
> 
> https://github.com/bmax121/APatch/actions/runs/8174234942/artifacts/1302558629

This version doesn't recognize my super key, unable to test.

bmax121, 2024-03-07:
fixed in main branch now

closed 2024-03-07 by bmax121

TheSingular, 2024-03-07:
Thanks for the fix, both of you!