bmax121/APatch #208: Apatch closes unexpectedly while goes AndroidPatch installation

SFMFnafer, 2024-02-17:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English. 请使用英文，不然会被关闭。
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

After flashing patched boot.img and rebooting into the system I opened Apatch and started installation of AndroidPatch. But after a while it's just closes unexpectedly.

### Expected behavior

I was expected an installed AndroidPatch and root rights working

### Device info

- Device: Poco X3 Pro
- OS Version: HyperOS 1.0.24.1.22.DEV
- APatch Version: Latest at the moment (10472)
- Kernel Version: 4.14.190
- KernelPatch Version: 0.8.5

SFMFnafer, 2024-02-17:
> ### Please check before submitting an issue
> * [x]  I have searched the issues and haven't found anything relevant
> * [x]  If patch failed, root failed, or device unable to boot after flashing the new [...]

[APatch_bugreport_2024-02-17_20_14.tar.gz](https://github.com/bmax121/APatch/files/14319691/APatch_bugreport_2024-02-17_20_14.tar.gz)

SFMFnafer, 2024-02-17:
attached the logs file

bmax121, 2024-03-08:
Try the latest version, and if it still doesn't work, reopen the issue.

closed 2024-03-08 by bmax121