bmax121/APatch #227: 给 MacroDroid 授权 root 后，MacroDroid 白屏崩溃，无法使用。

m1mark, 2024-02-27:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English.
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

给 MacroDroid 应用授予 Root 权限后，MacroDroid 无法正常打开，白屏崩溃。关闭 root 权限，应用正常。

After granting Root permissions to the MacroDroid app, MacroDroid fails to open properly, resulting in a white screen crash. The app functions normally when Root permissions are revoked.

### Logs

[APatch_bugreport_2024-02-27_20_26.tar.gz](https://github.com/bmax121/APatch/files/14419478/APatch_bugreport_2024-02-27_20_26.tar.gz)

### Device info

- Device:小米 11 pro
- OS Version:14.0.11.0.TKACNXM
- APatch Version:版本： 
10472-2-gb56bd4b (10474)
版本：0.8.5
- Kernel Version:5.4.210-qgki-g71d7658432ee
- KernelPatch Version:版本： 
10472-2-gb56bd4b (10474)

pomelohan, 2024-02-27:
Edit your issue to make it described in English please.

marat2509, 2024-02-28:
> * My problem is fully stated in English

Mention me when you actually done this task.

TheSingular, 2024-03-02:
> > * My problem is fully stated in English
> 
> Mention me when you actually done this task.

@marat2509 I think they did it but failed to mention you, so I'll do it in their stead.
The only part not fully translated is the device info part, device is apparently Xiaomi 11 pro.

bmax121, 2024-03-08:
Try the latest version, and if it still doesn't work, reopen the issue.

closed 2024-03-08 by bmax121