bmax121/APatch #256: an app can't keep root permission after reboot

lingfengyu-dreaming, 2024-03-07:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English.
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

When I give root for [freezer](https://github.com/Nep-Timeline/Freezer) in apatch, it can work with root, but when I reboot my device, the permission won't keep, and I need to enable root for it again.

### To Reproduce

- 1. enable root for freezer in apatch
- 2. reboot device
- 3. open apatch and check the root permission for freezer

### Expected behavior

reboot device and keep the root permission for freezer

### Logs

[APatch_bugreport_2024-03-07_21_21.tar.gz](https://github.com/bmax121/APatch/files/14524631/APatch_bugreport_2024-03-07_21_21.tar.gz)

### Device info

- Device: OnePlus ACE2V
- OS Version: ColorOS 14.0.500
- APatch Version: 10474 & 10529
- Kernel Version: 5.10.168-android12-9
- KernelPatch Version: 0.10.2

bmax121, 2024-03-08:
This issue occurs when you uninstall or install anew; Try the latest version, closing permissions should prevent it from happening. and if it still doesn't work, reopen the issue.

closed 2024-03-08 by bmax121

lingfengyu-dreaming, 2024-03-08:
fixed

Skorpion96, 2024-07-17:
This is so dumb..... It happens because on /data/adb/ap/package_config there is still the old uid, so if you edit the file and put the new uid root will be granted. Maybe a sort of fix would be appreciated as it's annoying but it's a workaround I guess

OpM3TA, 2024-12-13:
> This issue occurs when you uninstall or install anew; Try the latest version, closing permissions should prevent it from happening. and if it still doesn't work, reopen the issue.

Can someone put this is English ffs. I have app that can't keep privileges annoying.

burntoc, 2025-01-24:
Yeah, this is still an issue with some apps  - hopw is this "fixed"?