bmax121/APatch #259: Can't run 'su' from Termux

MrLumin06, 2024-03-07:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English.
- [ ] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

I'm experiencing problems where in termux when trying to use the root shell, it returns the binary isn't found. I did not have such problems in the previous version. In other terminal emulators access to the root shell works.

### To Reproduce

1. Install Termux
2. Grant it's superuser permissions in the manager
3. Open the app
4. Wait until download and install bootstrap packages
5. Run 'su' command

### Expected behavior

You should have access to the root shell.

### Screenshots

[image] [image]

### Logs

[APatch_bugreport_2024-03-07_18_13.tar.gz](https://github.com/bmax121/APatch/files/14527698/APatch_bugreport_2024-03-07_18_13.tar.gz)

### Device info

- Device: Xiaomi Redmi Note 11 NFC (spesn)
- OS Version: HyperOS by Xiaomi.eu (OS1.0.3.0 TGKMIXM)
- APatch Version: 10529
- Kernel Version: 4.19.157-perf-g8a3ae9055596
- KernelPatch Version: 0.10.2

Admirepowered, 2024-03-08:
try /system/bin/su

bmax121, 2024-03-08:
fixed #264

closed 2024-03-08 by bmax121