bmax121/APatch #260: Su not work at termux but su work at root explorer

Siarrrrrrrr, 2024-03-07:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English.
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

[APatch_bugreport_2024-03-08_00_38.tar.gz](https://github.com/bmax121/APatch/files/14528043/APatch_bugreport_2024-03-08_00_38.tar.gz)
This report log

This is video for example
https://pixeldrain.com/u/2goYLRBX

### Device info

- Device: Fog (Reedmi 10c)
- OS Version: 13 Custom Rom Rising OS
- APatch Version: 10529
- Kernel Version: 4.19.175
- KernelPatch Version: 0.10.2

1fvyuio333, 2024-03-08:
me too,But  run "/system/su" success.

chase535, 2024-03-08:
I'm not sure this is a new feature, but in theory, it does enhance Apatch's hiding ability. Now a feasible solution is add a new line `alias su="/system/bin/su"` in file  `/data/user/0/com.termux/files/usr/etc/bash.bashrc`

bmax121, 2024-03-08:
fixed #264

closed 2024-03-08 by bmax121

Siarrrrrrrr, 2024-03-08:
Thnks