bmax121/APatch #266: The way the super key is stored is not secure enough./超級密鑰存储方式不够安全

RejectVanity, 2024-03-08:
I hope that only one manager can communicate with "Supercall" at a time. This will effectively prevent malware from hijacking control.

I also hope that the key is encrypted or not stored directly (the key is cleared directly after the authentication with Supercall is complete). Storing the plain text key in the manager is extremely insecure.

我希望同时只有一个管理器能与“Supercall”通信，这将有效防止被恶意软件篡夺管理权。

我还希望对密钥进行加密存储或者直接不存储（与Supercall通信认证完毕后直接清除密钥），密钥明文储存于管理器中，是极度不安全的。
把它翻译为英文

bmax121, 2024-03-09:
why is extremely insecure.

pomelohan, 2024-03-09:
I don't agree that it is extremely insecure unless your password contains your personal information such as birthday or name, or is the same as your password on other platforms, but there is a possibility of being exploited by malicious people (such as someone wanting to secretly collect your password without your permission and upload it to your server, marking it based on your local social account). From this perspective, there is indeed a certain risk, but it is not extremely insecure.

But we can consider adding an option to not store the auth key locally, so that users with requirements can open it themselves in the settings.

pomelohan, 2024-03-09:
> Why is extremely insecure, which scenarios . Actually, I haven't thought of a good way to save it yet.

How about making password hashed?

RejectVanity, 2024-03-09:
> Why is extremely insecure, which scenarios . Actually, I haven't thought of a good way to save it yet.

I'm not sure if a malicious app can gain root privileges on a device that has a manager installed, given the management rights for **KernelPatch** and **Root**. If not, it's not that dangerous.
我不清楚对于**KernelPatch**和**Root**的管理权，在**装置安装有管理器的情况下其它恶意App能否成功提权**，若不能则没那么危险。

bmax121, 2024-03-09:
The security level has indeed been reduced. It's illogical for an app with lower permissions to protect keys with higher privileges, but I have no immediate solution. For now, you can manually clear the keys or uninstall the manager.