bmax121/APatch #267: myandroidtools授权失败

Fabali, 2024-03-08:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English.
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

myandroidtools app在apatch中已经勾选授权，但是实际打开还是提示没有root权限

### Device info

- Device: 一加11
- OS Version: coloros14
- APatch Version: 10533
- Kernel Version: 5.15.94
- KernelPatch Version:0.10.3

bugme2, 2024-03-09:
juicessh也一样，最新版还是老毛病

bugme2, 2024-03-09:
juicessh也一样，10533最新版还是老毛病

ask9027, 2024-03-12:
also there is no root permission while root granted to Mixplorer

Typhonling, 2024-03-20:
还在用MyAndroidTools的我建议可以尝试一下[https://github.com/lihenggui/blocker](url)

Admirepowered, 2024-04-09:
Try ci version

Admirepowered, 2024-04-09:
https://github.com/bmax121/APatch/actions/runs/8617067391

pomelohan, 2024-04-10:
Fixed on https://github.com/bmax121/APatch/commit/064f079eccfa852f17ca22fdfac8921635ea296d

closed 2024-04-10 by pomelohan