bmax121/APatch #271: Apatch patching doesnt working

Datokina, 2024-03-09:
[image]
When I patch leaf os latest version this happens

Datokina, 2024-03-09:
Here's the log: 
[APatch_bugreport_2024-03-09_14_51.tar.gz](https://github.com/bmax121/APatch/files/14546149/APatch_bugreport_2024-03-09_14_51.tar.gz)

pomelohan, 2024-03-09:
Provide kernel please

UnboundVanila, 2024-03-10:
Just had the same issue

pomelohan, 2024-03-10:
> Just had the same issue

Provide Kernel Here

sekaiacg, 2024-03-11:
@Datokina Please upload the kernel

UnboundVanila, 2024-03-11:
> > Just had the same issue
> 
> Provide Kernel Here

4.14.336-g38336f095d6f
Note: I contacted the rom developer and he said it's bcuz upstream breaks apatch

bmax121, 2024-03-11:
@UnboundVanila @Datokina No kernel image, no fix

pomelohan, 2024-03-11:
You can reopen the issue once upload kernel image.

Datokina, 2024-03-11:
I am really sorry for not responding, I had some issues anyway here's a similar kernel that refuses to be patched.
[boot.zip](https://github.com/bmax121/APatch/files/14564511/boot.zip)

Datokina, 2024-03-12:
> You can reopen the issue once upload kernel image.

Can this issue be reopened again?

sekaiacg, 2024-03-12:
This problem cannot be reproduced on Linux/other Android devices, please wait.

naokoshoto, 2024-03-19:
Had same issue, unable to patch boot img and showing exact error

Datokina, 2024-03-20:
> Had same issue, unable to patch boot img and showing exact error

I think @UnboundVanila is right; upstream kernel is the problem.

Skorpion96, 2024-03-26:
[boot.zip](https://github.com/bmax121/APatch/files/14761537/boot.zip)
Same issue, SM-T220 Android 14

w3023185276, 2024-03-28:
I do have the  same issue,
[image]
[Uploading boot.zip…]()

sekaiacg, 2024-03-28:
> I do have the same issue,

能提供远程调试吗，todesk

dickta, 2024-06-29:
I have the same issue on LineageOS 21, lineage-21.0-20240626-nightly-alioth-signed

closed 2024-08-03 by marat2509