bmax121/APatch #274: How can I flash the patched boot.img where in samsung encrypted devices

UnboundVanila, 2024-03-09:
I am using a custom rom and it doesn't work with twrp and as you know samsung devices don't have fastboot mode so how I flashthe patched boot.img?

pomelohan, 2024-03-10:
You can try to root your device with userspace-root like Magisk first, then use the command to flash boot img:
`su
dd if=<your patched image> of=/dev/block/by-name/<your boot partition>`

But we're not recommended you to do this because the patched boot image is not guaranteed being able to boot, if the patched image failed to boot, the only way to fix is by Odin or Qualcomm 9008.

UnboundVanila, 2024-03-10:
> You can try to root your device with userspace-root like Magisk first, then use the command to flash boot img: `su -c "dd if=<your patched image> of=/dev/block/by-name/<your boot partition>"`
> 
> But we're not recommended you to do this because the patched boot image is not guaranteed being able to boot, if the patched image failed to boot, the only way to fix is by Odin or Qualcomm 9008.

What is userspace-root ? and can I just compress the patched boot.img to a zip and flash it using adb sideload?

pomelohan, 2024-03-10:
1. Userspace root means root by modifying ramdisk like Magisk.
2. Of course you can, but you should write a script by yourself. Don’t forget to backup.

UnboundVanila, 2024-03-10:
I am really sorry if i am annoying you but how can I write a script? I am a newbie

UnboundVanila, 2024-03-10:
> 1. Userspace root means root by modifying ramdisk like Magisk.
> 2. Of course you can, but you should write a script by yourself. Don’t forget to backup.

I am really sorry but how can I write a script i am a newbie

pomelohan, 2024-03-10:
> > 1. Userspace root means root by modifying ramdisk like Magisk.
> > 2. Of course you can, but you should write a script by yourself. Don’t forget to backup.
> 
> I am really sorry but how can I write a script i am a newbie

I'm sorry but I don't have enough time to guide you to do this. You can ask ChatGPT or someone else to help you with this.
btw, why not try fastbootd with TWRP?

UnboundVanila, 2024-03-10:
> I'm sorry but I don't have enough time to guide you to do this. You can ask ChatGPT or someone else to help you with this. btw, why not try fastbootd with TWRP?

1.The twrp version of my phone only supports very few number of roms, most roms support their own recoveries
2. When you say "fastbootd in twrp" do you mean installing it directly? 
( I am gonna try making my own script and will update you with the result, Thank you so much for responding to my comments)

pomelohan, 2024-03-10:
`fastbootd in twrp` means flash boot partition by userspace fastboot, it was powered by TWRP, but you can use `fastboot flash boot xxxx` command like in bootloader. For more infomation, just Google it.

closed 2024-03-10 by pomelohan

Kyogre, 2024-10-19:
I just flashed via Odin: put patched `boot.img` into tar archive, then put tar in AP-slot in Odin.