bmax121/APatch #275: APatchLite

jerome-antoine, 2024-03-10:
Hi,
sorry, I know this is not an issue but could you please explain me what is/are the difference/s between APatch and APatchLite ? I have tried the latter for a few hours and apparently it doesn't have any KPModule, unlike APatch, but is there anything more that differs ?
I'd really appreciate to be enlightened on that matter.
Thank you.

pomelohan, 2024-03-10:
The APatch Lite is also an open source project. You can check the difference by yourself.

Here is the repository link: https://github.com/ponces/APatchLite

closed 2024-03-10 by pomelohan

jerome-antoine, 2024-03-10:
Thanks,
as I said, I have tried APatchLite for a few hours. As I am far from being an expert in APatch, I just wanted to know if there were other differences than the KPModule which is available in APatch, but absent in APatchLite. Of course I have checked the repository that you mentioned, since this is where I got my APK, but it doesn't explain anything. Apart from the releases, it's just a copy/paste of the APatch repository.
Sorry if my question seems stupid or obvious, I don't intend to bother anybody.

pomelohan, 2024-03-10:
Your question is not stupid and has not disturbed anyone. Please don't think so and underestimate yourself too much, brother. The APatch community is very friendly and harmonious, and any questions or discussions related to the project are welcome. Closing the issue is simply because it is not a bug or requesting new features. If you want to continue discussing this issue, you can go to the discussions section of the project to continue the discussion. ;)
🌹Thanks for supporting APatch

jerome-antoine, 2024-03-10:
Thanks, I appreciate 👍