bmax121/APatch #279: 修复bug

2825817398, 2024-03-10:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English.
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

After the update, there was no response when entering the key, and worms appeared

### Expected behavior

希望修复

### Screenshots

[image]

### Logs

[APatch_bugreport_2024-03-10_02_12.tar.gz](https://github.com/bmax121/APatch/files/14548894/APatch_bugreport_2024-03-10_02_12.tar.gz)

### Device info

[image]

### Additional context

.

Admirepowered, 2024-03-10:
OTA update will flash boot image,lose su is normal

closed 2024-03-10 by Admirepowered

2825817398, 2024-03-10:
> OTA更新会闪存开机映像，丢失su是正常的

请问怎么解决

Admirepowered, 2024-03-10:
> > OTA更新会闪存开机映像，丢失su是正常的
> 
> 请问怎么解决

等待feature提供刷入另一卡槽的功能或者使用第三方软件在更新前修补镜像

Admirepowered, 2024-03-10:
ref: #265

pomelohan, 2024-03-10:
目前暂不支持直接通过 Manager OTA. 请等待后续版本支持，目前你可以再次手动修补后刷入

2825817398, 2024-03-10:
> 目前暂不支持直接通过 Manager OTA. 请等待后续版本支持，目前你可以再次手动修补后刷入

已解决感谢，感谢各位的指导！