bmax121/APatch #283: Unable to load the superuser list 无法加载超级用户列表

Mxiier, 2024-03-11:
### Please check before submitting an issue

- [X] I have searched the issues and haven't found anything relevant
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch
- [X] I will upload bugreport file in APatch Manager - Settings - Report log
- [X] My problem is fully stated in English.
- [X] I know how to reproduce the issue which may not be specific to my device

### Describe the bug

Unable to load the superuser list
[image]
[image]

### Reproduce method

Unable to load the superuser list 

### Expected behavior

Unable to load the superuser list 

### Actual behaviour

Unable to load the superuser list 

### Logs

[APatch_bugreport_2024-03-11_09_06.tar.gz](https://github.com/bmax121/APatch/files/14552955/APatch_bugreport_2024-03-11_09_06.tar.gz)

### Device Name

moto G51

### OS Version

android 11

### APatch Version

10533

### Kernel Version

5.4.86

### KernelPatch Version

0.10.3

### Additional context

[APatch_bugreport_2024-03-11_09_06.tar.gz](https://github.com/bmax121/APatch/files/14552963/APatch_bugreport_2024-03-11_09_06.tar.gz)

Mxiier, 2024-03-11:
This is Kernel
[kernel.zip](https://github.com/bmax121/APatch/files/14552991/kernel.zip)

pomelohan, 2024-03-11:
` D AndroidRuntime: Calling main entry com.topjohnwu.superuser.internal.RootServerMain
 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x30a89eb87c in tid 10475 (main), pid 10475 (main)
 I crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
 I tombstoned: received crash request for pid 10475
 I crash_dump64: performing dump of process 10475 (target tid = 10475)
 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
 F DEBUG   : Build fingerprint: 'motorola/cypfq_cmcc/cypfq:11/RRYA31.Q3-56-99-2/0e9209:user/release-keys'
 F DEBUG   : Revision: 'pvt'
 F DEBUG   : ABI: 'arm64'
 F DEBUG   : Timestamp: 2024-03-11 09:05:19+0800
 F DEBUG   : pid: 10475, tid: 10475, name: main  >>> me.bmax.apatch:root:0 <<<
 F DEBUG   : uid: 0
 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x30a89eb87c
 F DEBUG   :     x0  0000006e1ee54010  x1  0000000000000050  x2  000000006ff77708  x3  0000007fdb8e52f0
 F DEBUG   :     x4  0000000000000000  x5  b400006e3d5f1214  x6  0000000000000031  x7  0000000000000031
 F DEBUG   :     x8  00000000fffffbff  x9  0000000000000018  x10 0000006e1ee54118  x11 0000000000000020
 F DEBUG   :     x12 0000000000000000  x13 0000007fdb8e54d0  x14 0000000000000002  x15 0000000000000000
 F DEBUG   :     x16 00000070bd8347d0  x17 00000070bec8bad4  x18 00000070c36fe000  x19 0000006e1ee54010
 F DEBUG   :     x20 00000030a89eb86c  x21 00000030a89eb86c  x22 000000000000000c  x23 0000007fdb8e52f0
 F DEBUG   :     x24 b400006f3d5e8548  x25 0000000000000060  x26 00000070c2e44000  x27 0000006e2ce8f614
 F DEBUG   :     x28 0000000000000001  x29 0000007fdb8e5190
 F DEBUG   :     lr  0000006e2d224cec  sp  0000007fdb8e5190  pc  0000006e2d10b388  pst 0000000000001000
 F DEBUG   : backtrace:
 F DEBUG   :       #00 pc 000000000023b388  /apex/com.android.art/lib64/libart.so (mspace_malloc+116) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #01 pc 0000000000354ce8  /apex/com.android.art/lib64/libart.so (art::jit::JitMemoryRegion::AllocateData(unsigned long)+20) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #02 pc 000000000034f680  /apex/com.android.art/lib64/libart.so (art::jit::JitCodeCache::AddProfilingInfoInternal(art::Thread*, art::ArtMethod*, std::__1::vector<unsigned int, std::__1::allocator<unsigned int> > const&)+148) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #03 pc 000000000034f5ac  /apex/com.android.art/lib64/libart.so (art::jit::JitCodeCache::AddProfilingInfo(art::Thread*, art::ArtMethod*, std::__1::vector<unsigned int, std::__1::allocator<unsigned int> > const&, bool)+184) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #04 pc 0000000000354ff0  /apex/com.android.art/lib64/libart.so (art::ProfilingInfo::Create(art::Thread*, art::ArtMethod*, bool)+412) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #05 pc 0000000000343018  /apex/com.android.art/lib64/libart.so (art::jit::Jit::MaybeCompileMethod(art::Thread*, art::ArtMethod*, unsigned int, unsigned int, bool)+800) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #06 pc 0000000000319e78  /apex/com.android.art/lib64/libart.so (art::interpreter::ArtInterpreterToCompiledCodeBridge(art::Thread*, art::ArtMethod*, art::ShadowFrame*, unsigned short, art::JValue*)+588) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #07 pc 00000000003100d0  /apex/com.android.art/lib64/libart.so (bool art::interpreter::DoCall<false, false>(art::ArtMethod*, art::Thread*, art::ShadowFrame&, art::Instruction const*, unsigned short, art::JValue*)+996) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #08 pc 000000000068057c  /apex/com.android.art/lib64/libart.so (MterpInvokeInterface+1032) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #09 pc 000000000012da14  /apex/com.android.art/lib64/libart.so (mterp_op_invoke_interface+20) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #10 pc 000000000035074e  /system/framework/framework.jar (android.content.pm.PackageParser$ApkLite.<init>+114)
 F DEBUG   :       #11 pc 00000000003076cc  /apex/com.android.art/lib64/libart.so (art::interpreter::Execute(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame&, art::JValue, bool, bool) (.llvm.13239354378677934218)+268) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #12 pc 000000000066d8b4  /apex/com.android.art/lib64/libart.so (artQuickToInterpreterBridge+780) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #13 pc 000000000013cff8  /apex/com.android.art/lib64/libart.so (art_quick_to_interpreter_bridge+88) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
 F DEBUG   :       #14 pc 000000000036b394  /system/framework/arm64/boot-framework.oat (android.content.pm.PackageParser.parseApkLite+6468) (BuildId: a7a3a6efdb6d7afd7d720df0097dc1530b96232f)
 F DEBUG   :       #15 pc 000000000036b7c4  /system/framework/arm64/boot-framework.oat (android.content.pm.PackageParser.parseApkLiteInner+948) (BuildId: a7a3a6efdb6d7afd7d720df0097dc1530b96232f)
 F DEBUG   :       #16 pc 0000000000369a24  /system/framework/arm64/boot-framework.oat (android.content.pm.PackageParser.parseApkLite+52) (BuildId: a7a3a6efdb6d7afd7d720df0097dc1530b96232f)
 F DEBUG   :       #17 pc 00000000001337e8  /apex/com.android.art/lib64/libart.so (art_quick_invoke_static_stub+568) (BuildId: 5a57bef0c3bc3adef9f99a7e80bfa5eb)
[... cut ...]
 F DEBUG   :       #82 pc 0000000000003580  /system/bin/app_process64 (main+1336) (BuildId: fc3c6038109116dd58f5ab8044856fc7)
 F DEBUG   :       #83 pc 000000000004988c  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+108) (BuildId: 1fdafb7d457cc367eb58ced21a4fa8ac)
 W NativeCrashListener: Couldn't find ProcessRecord for pid 10475`
 
 在通过 RootService 获取应用列表的时候直接触发 art 崩溃了，看起来是联想魔改了什么东西导致的。暂时的工作重心不会放在这里，但在未来的某一天里可能会修复。如果可以的话希望可以提供 /system/framework/framework.jar 和 /system/framework/services.jar

Mxiier, 2024-03-12:
好的 
[framwork.tar.gz](https://github.com/bmax121/APatch/files/14570060/framwork.tar.gz)

pomelohan, 2024-03-13:
试试这个版本
https://github.com/bmax121/APatch/actions/runs/8266035875/artifacts/1322513027

Mxiier, 2024-03-14:
你好,非常感谢,我测试了下可以加载超级用户列表了，还测试了些其它功能，我发现 隐藏Apatch管理器 功能会闪退，无法成功隐藏，和安装magisk模块好像有问题。
如果有空希望解决下，谢谢
这是日志文件和录屏
[apatch-log.zip](https://github.com/bmax121/APatch/files/14596767/apatch-log.zip)

pomelohan, 2024-03-14:
> 你好,非常感谢,我测试了下可以加载超级用户列表了，还测试了些其它功能，我发现 隐藏Apatch管理器 功能会闪退，无法成功隐藏，和安装magisk模块好像有问题。 如果有空希望解决下，谢谢 这是日志文件和录屏 [apatch-log.zip](https://github.com/bmax121/APatch/files/14596767/apatch-log.zip)

因为是 topjohnwu 的 libsu 在 bind RootService 的时候出现的 library 崩溃，所以我换到了 pms 获取应用列表的方案，但是团队内部对于我的修改表示特别反对（据他们说是会导致部分设备无法获取应用列表、获取的不全等，但他们用的不是这个 API），因此这个修改暂时不能合并主线。但是我还是保留了这个修改在另外的分支：
https://github.com/bmax121/APatch/tree/query-applist-pms
我们这边也会增加测试的样本，来验证这个 API 的稳定性，如果验证确实稳定没有异常，则会合并到主线。

至于隐藏 APatch Manager 功能闪退，关键日志如下：
```
E AndroidRuntime: FATAL EXCEPTION: main  
E AndroidRuntime: Process: me.bmax.apatch, PID: 3714  
E AndroidRuntime: java.lang.NoSuchMethodError: No direct method <init>(Ljava/io/OutputStream;ZLjava/nio/charset/Charset;)V in class Ljava/io/PrintStream; or its super classes (declaration of 'java.io.PrintStream' appears in /apex/com.android.art/javalib/core-oj.jar)  
E AndroidRuntime: 	at m2.o(Unknown Source:3)  
E AndroidRuntime: 	at Aa0.c(Unknown Source:728)  
E AndroidRuntime: 	at PC.t(Unknown Source:243)  
E AndroidRuntime: 	at Pc.o(Unknown Source:5)  
E AndroidRuntime: 	at ls.run(Unknown Source:111)  
E AndroidRuntime: 	at X2.run(Unknown Source:37)  
E AndroidRuntime: 	at ig0.run(Unknown Source:2)  
E AndroidRuntime: 	at Dn.run(Unknown Source:91)  
E AndroidRuntime: 	Suppressed: qr: [iQ@34ceb54, Ne@e6bfafd, ed0{Cancelling}@72123f2, h7@13c3a43]  
```

不难看出你的系统实在是太糟糕了，目前暂时没有解决方案。

安装 APM 模块时出现 OS ERROR 22 是已知问题，不仅仅是你的系统你的设备出现了，后面会想办法修复，到时候可能还需要您的帮忙测试。

因为这个问题已经解决掉了，所以我将关闭这个 issue。关于安装 APM 模块时出现 OS ERROR 22，你可以再另行开一个 issue。

closed 2024-03-14 by pomelohan

nitanmarcel, 2024-03-15:
The issue seems to persist here

nitanmarcel, 2024-03-15:
But my issue seems different

https://bin.kv2.dev/~65f4075c5de7ed6bfdbfad3c

pomelohan, 2024-03-15:
> But my issue seems different
> 
> https://bin.kv2.dev/~65f4075c5de7ed6bfdbfad3c

Do you have similar issue when using other root required app which is using `topjohnwu's libsu` ?

We couldn't fix this since it was caused by topjohnwu's libsu. It would be better if you provide contexts to libsu's issue page.

nitanmarcel, 2024-03-15:
> Do you have similar issue when using other root required app which is using `topjohnwu's libsu` ?
>
> We couldn't fix this since it was caused by topjohnwu's libsu. It would be better if you provide [...]

Ah yes. For example [GMS Flags](https://github.com/polodarb/GMS-Flags) which uses libsu is unable to load any data.

Should I open an issue on libsu repository is what you mean by providing context to libsu's issue page?

nitanmarcel, 2024-03-15:
> Do you have similar issue when using other root required app which is using `topjohnwu's libsu` ?
>
> We couldn't fix this since it was caused by topjohnwu's libsu. It would be better if you provide [...]
I've opened an issue

https://github.com/bmax121/APatch/issues/283#issuecomment-1999868919

pomelohan, 2024-03-15:
> Ah yes. For example [GMS Flags](https://github.com/polodarb/GMS-Flags) which uses libsu is unable to load any data.
>
> Should I open an issue on libsu repository is what you mean by providing context [...]

I mean just open an issue on libsu instead.

nitanmarcel, 2024-03-16:
> I mean just open an issue on libsu instead.

Yeah, I opened one yesterday waiting for a reply to it at the moment.

What I find strange is that solutions like KernelSU doesn't have this issue. Are we sure there isn't something with the APatch implementation?

nitanmarcel, 2024-03-16:
Not even su in termux works, throws permission errors. Modules can't seem to install either. And I can't share the logs. I tried saving it with total commander and it complains that it can't open it for reading 

[APatch_install_log_2024-03-16-12-16-18.log](https://github.com/bmax121/APatch/files/14623213/APatch_install_log_2024-03-16-12-16-18.log)

nitanmarcel, 2024-03-16:
> I mean just open an issue on libsu instead.

I've also opened an issue here regarding other problems I have:

https://github.com/bmax121/APatch/issues/297