bmax121/APatch #290: Magisk like zip installer for flashing through recovery

elisenlebkuch, 2024-03-14:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

No.

### Describe the solution you'd like/描述你想要的解决方案

With Magisk, you can rename the installer from .apk to .zip and flash it via recovery.

### Describe alternatives you've considered/描述您考虑过的备选方案

.

pomelohan, 2024-03-14:
Yes, we can do that but how to handle SuperKey?

If the SuperKey is too simple, this would be a risk. Instead, too complex makes it too hard to remember. Also, we can’t set SuperKey by inputting customized one during flashing in Recovery. Do you have a good idea about this?

elisenlebkuch, 2024-03-14:
You could set it afterwards in the app and flash it again from there as you now have root priveleges.

nitanmarcel, 2024-03-15:
> Yes, we can do that but how to handle SuperKey?
> 
> If the SuperKey is too simple, this would be a risk. Instead, too complex makes it too hard to remember. Also, we can’t set SuperKey by inputting customized one during flashing in Recovery. Do you have a good idea about this?

simply just ask the user to do `echo {superkey} >> /dev/superkey` or any readable location from recovery and automatically delete it after

pomelohan, 2024-03-15:
> You could set it afterwards in the app and flash it again from there as you now have root priveleges.

I thought this maybe a good idea. We can call it as temporarily password. After booting, we should force resetting the password.

pomelohan, 2024-03-15:
> simply just ask the user to do `echo {superkey} >> /dev/superkey` or any readable location from recovery and automatically delete it after

Maybe it's too complex for beginners.

pomelohan, 2024-03-15:
However, there is still a risk of unable to boot after patching (even successfully patched) or patching failed. I'm waiting for team's opinion about this.

ttimasdf, 2024-03-26:
Generating a random complicated superkey and prompt user to backup the password on app first launch, solves this problem, and also provides a better default than letting user to choose their own one. Because users will likely choose weak passwords which is used only once.

ttimasdf, 2024-03-26:
Another way is to set super key by installer zip file name. like `Apatch.[supersecret].apk`

like Magisk, which run uninstaller script instead of install/update when apk name contains `uninstall`

https://github.com/topjohnwu/Magisk/blob/21ed09560184abe18e213b890a9b8950181b1d77/scripts/update_binary.sh#L23-L24

fsxitutu, 2024-07-31:
The average user doesn't use too many operations that require a superkey, and usually doesn't move the superkey after entering it into the apatch manager. So it is possible to generate a strong superkey and store the superkey in a file.
Based on the above idea, can we provide a stub apk, using the same package name and signature of apatch but only providing the function of downloading and installing apatch apk (similar to magisk-stud). Encapsulate the stub apk and superkey file into ramdisk, install the stub apk after booting and write the superkey to the apatch data directory (/data/data/me.bmax.apatch), and then the user installs the apatch and then the superkey is read by the apatch, to achieve the effect of using the apatch out-of-the-box!
For example, magisk, in the recovery process, it will use xz compression to package the stub apk to ramdisk, and then automatically install the stub apk after booting, which provides a convenient way for users to download the apk.

Translated with DeepL.com (free version)
English is not my native language, the English content is translated using translation software, please understand and point out if it is not correct. Here is the original Chinese text :
一般用户并不会使用过多需要superkey的操作，通常往apatch管理器输入superkey后就不会再动superkey了。所以完全可以生成强superkey后把superkey存储到文件中。
基于上述观点，能否提供一个stud apk，使用apatch相同的包名和签名但仅提供下载安装apatch apk的功能(类似于magisk-stud)。将stud apk和superkey文件封装到ramdisk中，开机后安装stud apk并将superkey写入到apatch数据目录(/data/data/me.bmax.apatch)，用户安装apatch后由apatch读取superkey，达到开箱即用的效果
比如magisk，在recovery刷入过程中，会将stud apk使用xz压缩后封装到ramdisk，开机后自动安装stud apk，为用户提供一个方便的下载途径

SakuraKyuo, 2024-08-03:
As I did, I added the flash script to APatch in https://github.com/bmax121/APatch/pull/581, and I also tried to make it work in all recovery, including AOSP-based and TWRP-based recovery(https://github.com/bmax121/APatch/pull/583).

Regarding the solution for SuperKey, I will generate a random 8-digit number and letter combination as SuperKey, and if the user forgets it, it will be impossible to get it again, unless APatch leaks the plaintext key.

closed 2024-08-04 by pomelohan