bmax121/APatch #293: Unable not flashing magisk modules and Hidden apatch manager. 无法刷入magisk模块和隐藏apatch管理器

Mxiier, 2024-03-15:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Unable not flashing magisk modules Hidden apatch manager. 无法刷入magisk模块和隐藏apatch管理器

### Reproduce method/复现方法

Unable not flashing magisk modules. 无法刷入magisk模块和隐藏apatch管理器

### Expected behavior/预期行为

Unable not flashing magisk modules and Hidden apatch manager. 无法刷入magisk模块和隐藏apatch管理器

### Actual behaviour /实际行为

Unable not flashing magisk modules and Hidden apatch manager. 无法刷入magisk模块和隐藏apatch管理器

### Screenshots/截图

[image]

Uploading screen-20240314-121058.mp4…

### Logs/日志

[APatch_bugreport_2024-03-14_12_39.tar.gz](https://github.com/bmax121/APatch/files/14610695/APatch_bugreport_2024-03-14_12_39.tar.gz)
[javalib.zip](https://github.com/bmax121/APatch/files/14610704/javalib.zip)

### Device Name/设备名称

moto G51

### OS Version/系统版本

myui 2.0 (android 11)

### APatch Version/APatch 版本

10533-10566

### Kernel Version/内核版本

5.4.86

### KernelPatch Version/KernelPatch 版本

0.10.3

### Additional context/其他信息

[APatch_bugreport_2024-03-14_12_39.tar.gz](https://github.com/bmax121/APatch/files/14610722/APatch_bugreport_2024-03-14_12_39.tar.gz)
[javalib.zip](https://github.com/bmax121/APatch/files/14610724/javalib.zip)

bmax121, 2024-03-17:
能在开机后立刻运行一下 dmesg 吗，运气好的话能拿到日志

Mxiier, 2024-03-18:
您好,我试了一下需要重启两次才可以正常刷入模块，但使用隐藏Apatch管理器功能依然会闪退。
在 logcat 中的提示是 java.lang.NoSuchMethodErrorjava.lang.NoSuchMethodError: No direct method <init>(Ljava/io/OutputStream;ZLjava/nio/charset/Charset;)V in class Ljava/io/PrintStream; or its super classes (declaration of 'java.io.PrintStream' appears in /apex/com.android.art/javalib/core-oj.jar)

bmax121, 2024-03-18:
安装 AndroidPatch 但是没重启是没有权限的，需要改一下，获取提示重启一下

pomelohan, 2024-03-24:
> 您好,我试了一下需要重启两次才可以正常刷入模块，但使用隐藏Apatch管理器功能依然会闪退。 在 logcat 中的提示是 java.lang.NoSuchMethodErrorjava.lang.NoSuchMethodError: No direct method (Ljava/io/OutputStream;ZLjava/nio/charset/Charset;)V in class Ljava/io/PrintStream; or its super classes (declaration of 'java.io.PrintStream' appears in /apex/com.android.art/javalib/core-oj.jar)

隐藏管理器功能后期可能会下线/重新实现，暂时不是工作中心。模块问题可以试试 10570 之后的版本，如果没有问题请关闭 issues.

pomelohan, 2024-03-30:
因为长时间没有回复，本 issue 暂时关闭，如仍有问题可以重新开启。

closed 2024-03-30 by pomelohan