bmax121/APatch #297: Apatch broken on Android 14.

nitanmarcel, 2024-03-16:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Apatch is completly broken on by device running Android 14 on crdroid.

The issue range from libsu not properly working to modules not being installed. Exporting the apatch logs does not work either.

### Reproduce method/复现方法

1. Install and patch the kernel
2. Open Apatch, and try to install modules or grant su permissions (needs the active PR)
3. Try to use su in termux

### Expected behavior/预期行为

All of the above should work

### Actual behaviour /实际行为

Everything is broken

### Logs/日志

Logcat: 
[log.txt](https://github.com/bmax121/APatch/files/14623376/log.txt)

KernelConfig: 
[config.txt](https://github.com/bmax121/APatch/files/14623383/config.txt)

### Device Name/设备名称

Oneplus 6t

### OS Version/系统版本

14 (crDroid)

### APatch Version/APatch 版本

10568 - PR #289

### Kernel Version/内核版本

4.9.337

### KernelPatch Version/KernelPatch 版本

0.10.3

### Additional context/其他信息

1. I've opened an issue on the [libsu's issue page](https://github.com/topjohnwu/libsu/issues/179) regarding the problem with their library.
2. Modules can't be installed, instead I'm granted with a blank page, and while exporting the installation log I get only a line saying `Error: No such file or directory (os error 2)`.
3. I can't export the APatch logs. I tried sending it to me via email, telegram, but to no avail. Also using Total Comander to try and save it to my storage, it complains that it can't open it for reading.
4. Su in termux is also broken, while I can enter the su shell I get permission errors when trying to access any directories

bmax121, 2024-03-17:
can you get root shell in command line

nitanmarcel, 2024-03-17:
> can you get root shell in command line

I can enter root shell. But it only works in adb. Ir termux if I try to access a root directory I get permission errors.

nitanmarcel, 2024-03-17:
[image]

pomelohan, 2024-03-17:
try latest version

nitanmarcel, 2024-03-18:
> try latest version

Same issues on the latest version

nitanmarcel, 2024-03-18:
> Im using 0.10.4 and A14 kernel 4.9.337, it is working..

Weird. Maybe there's something in my kernel that breaks it but no idea what. Mind you share your config?

`zcat /proc/config.gz`

nitanmarcel, 2024-03-18:
> /data/adb/ap/log

Yes

https://gofile.io/d/5dRG7Y

nitanmarcel, 2024-03-21:
Live patching the kernel is broken too

[image]

nitanmarcel, 2024-03-21:
Looks like it was a kernel/rom issue. No idea what caused it but I switched to LineageOS and everything works fine now.

closed 2024-03-21 by nitanmarcel