bmax121/APatch #299: Make Super Key authorization through fingerprint

muhammadbahaa2001, 2024-03-17:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

No

### Describe the solution you'd like/描述你想要的解决方案

Just put your fingerprint for Super Key authorization instead of typing password as in Magisk.

### Describe alternatives you've considered/描述您考虑过的备选方案

No alternatives

pomelohan, 2024-03-17:
Onky the first time opening APatch requires SuperKey. Then, it will automatically load encrypted password from local the next time APatxh starts. Which situation do we need this feature?

muhammadbahaa2001, 2024-03-18:
> Onky the first time opening APatch requires SuperKey. Then, it will automatically load encrypted password from local the next time APatxh starts. Which situation do we need this feature?

I meant super key should be used for granting root access from the manager, but with fingerprint instead of using password.

closed 2024-03-19 by bmax121