bmax121/APatch #304: Termux terminal lacking selinux permissions

sabpprook, 2024-03-18:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

```
Welcome to Termux!

Community forum: https://termux.com/community
Gitter chat:     https://gitter.im/termux/termux
IRC channel:     #termux on libera.chat

Working with packages:

 * Search packages:   pkg search <query>
 * Install a package: pkg install <package>
 * Upgrade packages:  pkg upgrade

Subscribing to additional repositories:

 * Root:     pkg install root-repo
 * X11:      pkg install x11-repo

Report issues at https://termux.com/issues

~ $ su
:/data/data/com.termux/files/home # cd /
:/ # id
uid=0(root) gid=0(root) groups=0(root) context=u:r:magisk:s0
:/ # pm list packages
cmd: Failure calling service package: Failed transaction (2147483646)
2|:/ # logcat | grep avc
03-19 06:14:00.396  1819  1819 W /vendor/bin/hw/vendor.qti.hardware.perf-hal-service: type=1400 audit(0.0:20773): avc:  denied  { search } for  comm=41646170744C61756E636820566D name="15891" dev="proc" ino=6799647 scontext=u:r:vendor_hal_perf_default:s0 tcontext=u:r:magisk:s0 tclass=dir permissive=0
03-19 06:14:00.400  1819  1819 W /vendor/bin/hw/vendor.qti.hardware.perf-hal-service: type=1400 audit(0.0:20774): avc:  denied  { search } for  comm=41646170744C61756E636820566D name="15891" dev="proc" ino=6799647 scontext=u:r:vendor_hal_perf_default:s0 tcontext=u:r:magisk:s0 tclass=dir permissive=0
03-19 06:14:00.612  1819  1819 W /vendor/bin/hw/vendor.qti.hardware.perf-hal-service: type=1400 audit(0.0:20775): avc:  denied  { search } for  comm=41646170744C61756E636820566D name="15891" dev="proc" ino=6799647 scontext=u:r:vendor_hal_perf_default:s0 tcontext=u:r:magisk:s0 tclass=dir permissive=0
03-19 06:14:00.616  1819  1819 W /vendor/bin/hw/vendor.qti.hardware.perf-hal-service: type=1400 audit(0.0:20776): avc:  denied  { search } for  comm=41646170744C61756E636820566D name="15891" dev="proc" ino=6799647 scontext=u:r:vendor_hal_perf_default:s0 tcontext=u:r:magisk:s0 tclass=dir permissive=0
03-19 06:14:00.828  1819  1819 W /vendor/bin/hw/vendor.qti.hardware.perf-hal-service: type=1400 audit(0.0:20778): avc:  denied  { search } for  comm=41646170744C61756E636820566D name="15891" dev="proc" ino=6799647 scontext=u:r:vendor_hal_perf_default:s0 tcontext=u:r:magisk:s0 tclass=dir permissive=0
03-19 06:14:05.300 30199 30199 W com.termux: type=1400 audit(0.0:20783): avc:  granted  { execute } for  name="login" dev="dm-85" ino=53556 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:05.312 18171 18171 W com.termux: type=1400 audit(0.0:20784): avc:  granted  { execute } for  name="login" dev="dm-85" ino=53556 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:05.312 18171 18171 W com.termux: type=1400 audit(0.0:20785): avc:  granted  { execute_no_trans } for  path="/data/data/com.termux/files/usr/bin/login" dev="dm-85" ino=53556 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:05.312 18171 18171 W com.termux: type=1400 audit(0.0:20786): avc:  granted  { execute } for  name="dash" dev="dm-85" ino=53532 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:05.332 18171 18171 W login   : type=1400 audit(0.0:20787): avc:  granted  { execute } for  path="/data/data/com.termux/files/usr/bin/dash" dev="dm-85" ino=53532 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:08.748 18202 18202 W bash    : type=1400 audit(0.0:20831): avc:  granted  { execute } for  name="su" dev="dm-85" ino=53392 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:08.752 18202 18202 W bash    : type=1400 audit(0.0:20832): avc:  granted  { execute } for  name="su" dev="dm-85" ino=53392 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:08.752 18202 18202 W bash    : type=1400 audit(0.0:20833): avc:  granted  { execute_no_trans } for  path="/data/data/com.termux/files/usr/bin/su" dev="dm-85" ino=53392 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:08.752 18202 18202 W bash    : type=1400 audit(0.0:20834): avc:  granted  { execute } for  name="dash" dev="dm-85" ino=53532 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:08.752 18202 18202 W su      : type=1400 audit(0.0:20835): avc:  granted  { execute } for  path="/data/data/com.termux/files/usr/bin/dash" dev="dm-85" ino=53532 scontext=u:r:untrusted_app_27:s0:c88,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c88,c257,c512,c768 tclass=file app=com.termux
03-19 06:14:23.128 18327 18327 W cmd     : type=1400 audit(0.0:20837): avc:  denied  { read write } for  path="/dev/pts/1" dev="devpts" ino=4 scontext=u:r:system_server:s0 tcontext=u:object_r:untrusted_app_all_devpts:s0:c88,c257,c512,c768 tclass=chr_file permissive=0
^C
130|:/ #
```

### Reproduce method/复现方法

run termux and enter the command below.
```
su
pm list packages
```

### Expected behavior/预期行为

it should execute system functions

### Actual behaviour /实际行为

denied by selinux

### Device Name/设备名称

OnePlus CPH2585

### OS Version/系统版本

Android 14

### APatch Version/APatch 版本

10570

### Kernel Version/内核版本

5.15.94-android13

### KernelPatch Version/KernelPatch 版本

0.10.4

### Additional context/其他信息

no

bmax121, 2024-03-19:
It's quite strange, I tried Termux 0.118, and even on an unmodified, unroot device, pm (not pm list packages) fails to run properly. I suspect this might be a bug in Termux.

Typhonling, 2024-03-20:
But I can't get logcat permission permanently even with Tasker. It uses _pm grant net.dinglisch.android.taskerm android.permission.READ_LOGS_ to get this permission.
With Apatch, Tasker needs to manually confirm each time when reading logcat, no problem with Magisk.

sabpprook, 2024-03-20:
> But I can't get logcat permission permanently even with Tasker. It uses `pm grant net.dinglisch.android.taskerm android.permission.READ_LOGS` to get this permission. With Apatch, Tasker needs to manually confirm each time when reading logcat, no problem with Magisk. #192

Could you fetch the log through adb shell by running ```su -c "logcat | grep avc:"``` and try logcat in termux terminal.

Typhonling, 2024-03-21:
> > But I can't get logcat permission permanently even with Tasker. It uses `pm grant net.dinglisch.android.taskerm android.permission.READ_LOGS` to get this permission. With Apatch, Tasker needs to manually confirm each time when reading logcat, no problem with Magisk. #192但即使使用 Tasker，我也无法永久获得 logcat 权限。 它使用 `pm grant net.dinglisch.android.taskerm android.permission.READ_LOGS` 来获取此权限。 使用Apatch，Tasker每次读取logcat时都需要手动确认，使用Magisk则没有问题。 第192章
> 
> Could you fetch the log through adb shell by running `su -c "logcat | grep avc:"` and try logcat in termux terminal.您可以通过运行 `su -c "logcat | grep avc:"` 通过 adb shell 获取日志并在 termux 终端中尝试 logcat 吗？

Sorry, I'm back to Magisk for now. no plans to continue testing until this issue is fixed.

pomelohan, 2024-04-03:
Fixed on 0876c5a

closed 2024-04-03 by pomelohan